coolsnowwolf/lede #13381: 使用lede tage下的20220716代码进行编译固件报错

tarryyun, 2025-03-07:
### 详细叙述

我编辑低版本的openwrt+固件版本是5.4的固件；用master最新的代码能够编译成功，是R25.2.5/Openwrt-23.05,这个版本刷在CR660X里面很卡，内存占用大；我在网上找到刷低版本的 R22.8.2 内核版本5.4.203的固件很好，但是没有我想要的插件，因此我想自己编译；
git clone --depth 1 --branch 20220716 https://github.com/coolsnowwolf/lede.git【分支源代码】

执行./scripts/feeds install -a后出现报错：
feeds/telephony/net/asterisk-chan-lantiq/Makefile: invalid group spec :vmmc=386

【vmm=386这个报错我没有管，直接执行的make V=s -j1 】

make[3]: Leaving directory '/home/tarry/lede/tools/libtool'
time: tools/libtool/compile#81.77#11.45#123.47
make[2]: Leaving directory '/home/tarry/lede'
make[1]: *** [tools/Makefile:158: /home/tarry/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyyyyynyynyyynynyyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/home/tarry/lede'
make: *** [/home/tarry/lede/include/toplevel.mk:230: world] Error 2

或者有什么办法用最新lede源master代码编译低版本的固件呢？

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

xiaomi CR6608

### 详细日志

feeds/telephony/net/asterisk-chan-lantiq/Makefile: invalid group spec :vmmc=386
【vmm=386这个报错我没有管，直接执行的make V=s -j1 】

make[3]: Leaving directory '/home/tarry/lede/tools/libtool'
time: tools/libtool/compile#81.77#11.45#123.47
make[2]: Leaving directory '/home/tarry/lede'
make[1]: *** [tools/Makefile:158: /home/tarry/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyyyyynyynyyynynyyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/home/tarry/lede'
make: *** [/home/tarry/lede/include/toplevel.mk:230: world] Error 2

yanxin152133, 2025-03-10:
https://github.com/coolsnowwolf/lede/issues/13349

tarryyun, 2025-03-10:
> [#13349](https://github.com/coolsnowwolf/lede/issues/13349)

我的文件已经解决：
这是因为在feeds.conf.default文件中添加了src-git telephony [https://github.com/coolsnowwolf/telephony.git，如果你的.config配置没有电话的需求，可以把这个软件源给删掉，我没有添加这个，目前来说，没有报错了](https://github.com/coolsnowwolf/telephony.git%EF%BC%8C%E5%A6%82%E6%9E%9C%E4%BD%A0%E7%9A%84.config%E9%85%8D%E7%BD%AE%E6%B2%A1%E6%9C%89%E7%94%B5%E8%AF%9D%E7%9A%84%E9%9C%80%E6%B1%82%EF%BC%8C%E5%8F%AF%E4%BB%A5%E6%8A%8A%E8%BF%99%E4%B8%AA%E8%BD%AF%E4%BB%B6%E6%BA%90%E7%BB%99%E5%88%A0%E6%8E%89%EF%BC%8C%E6%88%91%E6%B2%A1%E6%9C%89%E6%B7%BB%E5%8A%A0%E8%BF%99%E4%B8%AA%EF%BC%8C%E7%9B%AE%E5%89%8D%E6%9D%A5%E8%AF%B4%EF%BC%8C%E6%B2%A1%E6%9C%89%E6%8A%A5%E9%94%99%E4%BA%86)

closed 2025-03-14 by coolsnowwolf