coolsnowwolf/lede #13393: 极路由3Pro(C312B) 用最新源码编译，无法使用有线网口

fejich, 2025-03-12:
### 详细叙述

[image]

[image]

编译时候只是按自己需要勾选了 USB打印服务器 跟 msd_lite ，WireGuard 等几个常用 luci 插件。
没做其他修改。

手动 checkout 到 [2023](ramips: kernel 5.10: add mt7620-nand driver for NAND flash #11076) 源码后
https://github.com/coolsnowwolf/lede/pull/11076

编译出来的固件就能正常使用有线网口

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

极路由3Pro(C312B) 

### 详细日志

[内核日志.txt](https://github.com/user-attachments/files/19203682/default.txt)

[系统日志.txt](https://github.com/user-attachments/files/19203683/default.txt)

modelsun, 2025-03-14:
咋这么多交换机switch？

aiamadeus, 2025-03-14:
```
diff --git a/target/linux/ramips/dts/mt7620a_hiwifi_r33.dts b/target/linux/ramips/dts/mt7620a_hiwifi_r33.dts
index 2f8254e56..b2dd31c77 100644
--- a/target/linux/ramips/dts/mt7620a_hiwifi_r33.dts
+++ b/target/linux/ramips/dts/mt7620a_hiwifi_r33.dts
@@ -19,10 +19,9 @@
 	};
 
 	nand {
-		status = "okay";
+		compatible = "mtk,mt7620-nand";
 		#address-cells = <1>;
 		#size-cells = <1>;
-		compatible = "mtk,mt7620-nand";
 
 		partitions {
 			compatible = "fixed-partitions";
@@ -140,9 +139,8 @@
 	};
 
 	rtl8367rb {
-		compatible = "realtek,rtl8367b", "rtl8367b";
-		cpu_port = <6>;
-		realtek,extif1 = <1 0 1 1 1 1 1 1 2>;
+		compatible = "realtek,rtl8367b";
+		realtek,extif = <6 1 0 1 1 1 1 1 1 2>;
 		mii-bus = <&mdio0>;
 	};
 };
@@ -224,11 +222,11 @@
 };
 
 &bdinfo {
-        compatible = "nvmem-cells";
-        #address-cells = <1>;
-        #size-cells = <1>;
-
-        macaddr_bdinfo_18a: macaddr@18a {
-                reg = <0x18a 0x11>;
-        };
-};
\ No newline at end of file
+	compatible = "nvmem-cells";
+	#address-cells = <1>;
+	#size-cells = <1>;
+
+	macaddr_bdinfo_18a: macaddr@18a {
+		reg = <0x18a 0x11>;
+	};
+};

```
你试一下

fejich, 2025-03-17:
> ```
> diff --git a/target/linux/ramips/dts/mt7620a_hiwifi_r33.dts b/target/linux/ramips/dts/mt7620a_hiwifi_r33.dts
> index 2f8254e56..b2dd31c77 100644
> --- [...]

我机器刷了旧版，已经用做打印机服务器了。

后续看看其他有这机器的人再试试你的补丁

感谢你的回复

zfdx123, 2025-03-28:
https://github.com/coolsnowwolf/lede/commits/master/target/linux/generic/files/drivers/net/phy/rtl8367b.c

可能在提交过程中破坏了兼容性？是否每个端口都无法使用？？

fejich, 2025-03-29:
> https://github.com/coolsnowwolf/lede/commits/master/target/linux/generic/files/drivers/net/phy/rtl8367b.c
> 
> 可能在提交过程中破坏了兼容性？是否每个端口都无法使用？？

刷回原厂，然后再刷最新源码固件，尝试了两次都同样的问题。
所有有线网口都无法工作，WIFI 能连接。

最后 checkout 到最后 2023年4月4日 最后提交编译出来才正常

zfdx123, 2025-03-29:
我打开了一个问题 #13426 你有时间尝试，跟踪我之前说的这些提交的commit，来确定到哪个提交被破坏的吗，退回这一个文件相关代码即可，其他代码不动

zfdx123, 2025-03-29:
当然也有可能跟这些提交没有关系，只是在查找问题

aiamadeus, 2025-03-30:
> 刷回原厂，然后再刷最新源码固件，尝试了两次都同样的问题。

唉，你又不给日志。。。
```
[    5.325640] rtl8367b rtl8367rb: using MDIO bus 'mdio'
[    5.331553] rtl8367b rtl8367rb: RTL8367RB-VB chip found (num:6367 ver:0020)
[    6.402200] rtl8367b rtl8367rb: realtek,extif property is not defined or invalid
[    6.413816] rtl8367b rtl8367rb: chip setup failed, err=-22
[    6.419447] rtl8367b: probe of rtl8367rb failed with error -22
```
这个已经修了

fejich, 2025-03-31:
> 唉，你又不给日志。。。
>
> ```
> [    5.325640] rtl8367b rtl8367rb: using MDIO bus 'mdio'
> [    5.331553] rtl8367b rtl8367rb: RTL8367RB-VB chip found (num:6367 ver:0020)
> [    6.402200] rtl8367b rtl8367rb: [...]

我这个 极路由3Pro  刷了旧源码固件后，已经用作于公司的打印机服务器，不好拆下来再折腾。
只能等后续有这机器的其他人测试了。

zfdx123, 2025-05-04:
https://github.com/coolsnowwolf/lede/issues/13426#issuecomment-2849151251

经过测试，并没有问题，交换机运行良好

coolsnowwolf, 2026-09-05:
根据 issue 中给出的启动日志，故障是 rtl8367b 驱动不再接受旧的 `realtek,extif1` 属性；当前 LEDE 已将该 DTS 调整为 `realtek,extif`，并在 RTL8367B 驱动中保留兼容处理。后续评论也报告交换机已测试正常，因此确认已修复，关闭此 issue。

closed 2026-09-05 by coolsnowwolf