coolsnowwolf/lede #13411: 最新代码斐讯K3路由器固件编译失败

xzchsia, 2025-03-21:
### 详细叙述

最新代码K3编译失败，差不多自从3.13号之后编译就一直失败，每次编译失败的错误信息都相同。

2025-03-21T12:42:33.6965105Z make[8]: 'conf' is up to date.
2025-03-21T12:42:33.7047810Z boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
2025-03-21T12:42:33.7048265Z #
2025-03-21T12:42:33.7048499Z # configuration written to .config
2025-03-21T12:42:33.7048770Z #
2025-03-21T12:42:33.7128552Z Building backport-include/backport/autoconf.h ... done.
2025-03-21T12:42:34.3868961Z   CC [M]  /home/runner/work/Phicomm-K3-LEDE-Firmware-Lean/Phicomm-K3-LEDE-Firmware-Lean/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/backports-6.1.24/compat/backport-5.10.o
2025-03-21T12:42:34.6101205Z In file included from ./include/linux/kernel_stat.h:9,
2025-03-21T12:42:34.6101916Z                  from ./include/linux/cgroup.h:26,
2025-03-21T12:42:34.6102429Z                  from ./include/net/netprio_cgroup.h:11,
2025-03-21T12:42:34.6102958Z                  from ./include/linux/netdevice.h:42,
2025-03-21T12:42:34.6104952Z                  from /home/runner/work/Phicomm-K3-LEDE-Firmware-Lean/Phicomm-K3-LEDE-Firmware-Lean/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/backports-6.1.24/backport-include/linux/netdevice.h:3,
2025-03-21T12:42:34.6107606Z                  from /home/runner/work/Phicomm-K3-LEDE-Firmware-Lean/Phicomm-K3-LEDE-Firmware-Lean/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/backports-6.1.24/compat/backport-5.10.c:5:
2025-03-21T12:42:34.6111094Z /home/runner/work/Phicomm-K3-LEDE-Firmware-Lean/Phicomm-K3-LEDE-Firmware-Lean/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/backports-6.1.24/backport-include/linux/interrupt.h:49:1: error: static declaration of 'tasklet_setup' follows non-static declaration
2025-03-21T12:42:34.6113066Z    49 | tasklet_setup(struct tasklet_struct *t,
2025-03-21T12:42:34.6113490Z       | ^~~~~~~~~~~~~
2025-03-21T12:42:34.6115201Z In file included from /home/runner/work/Phicomm-K3-LEDE-Firmware-Lean/Phicomm-K3-LEDE-Firmware-Lean/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/backports-6.1.24/backport-include/linux/interrupt.h:4:
2025-03-21T12:42:34.6117451Z ./include/linux/interrupt.h:708:13: note: previous declaration of 'tasklet_setup' with type 'void(struct tasklet_struct *, void (*)(struct tasklet_struct *))'
2025-03-21T12:42:34.6118647Z   708 | extern void tasklet_setup(struct tasklet_struct *t,
2025-03-21T12:42:34.6119152Z       |             ^~~~~~~~~~~~~
2025-03-21T12:42:34.6661868Z make[9]: *** [scripts/Makefile.build:262: /home/runner/work/Phicomm-K3-LEDE-Firmware-Lean/Phicomm-K3-LEDE-Firmware-Lean/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/backports-6.1.24/compat/backport-5.10.o] Error 1
2025-03-21T12:42:34.6664042Z make[8]: *** [scripts/Makefile.build:497: /home/runner/work/Phicomm-K3-LEDE-Firmware-Lean/Phicomm-K3-LEDE-Firmware-Lean/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/backports-6.1.24/compat] Error 2
2025-03-21T12:42:34.6667564Z make[7]: *** [Makefile:1755: /home/runner/work/Phicomm-K3-LEDE-Firmware-Lean/Phicomm-K3-LEDE-Firmware-Lean/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/backports-6.1.24] Error 2
2025-03-21T12:42:34.6669315Z make[6]: *** [Makefile.build:13: modules] Error 2
2025-03-21T12:42:34.6669820Z make[5]: *** [Makefile.real:93: modules] Error 2
2025-03-21T12:42:34.6671310Z make[4]: *** [Makefile:121: modules] Error 2
2025-03-21T12:42:34.6672608Z make[4]: Leaving directory '/home/runner/work/Phicomm-K3-LEDE-Firmware-Lean/Phicomm-K3-LEDE-Firmware-Lean/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/backports-6.1.24'
2025-03-21T12:42:34.6674954Z make[3]: *** [Makefile:401: /home/runner/work/Phicomm-K3-LEDE-Firmware-Lean/Phicomm-K3-LEDE-Firmware-Lean/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/backports-6.1.24/.built] Error 2
2025-03-21T12:42:34.6676809Z make[3]: Leaving directory '/home/runner/work/Phicomm-K3-LEDE-Firmware-Lean/Phicomm-K3-LEDE-Firmware-Lean/openwrt/package/kernel/mac80211'
2025-03-21T12:42:34.6682867Z time: package/kernel/mac80211/regular/compile#0.81#0.56#1.32
2025-03-21T12:42:34.6691168Z     ERROR: package/kernel/mac80211 failed to build (build variant: regular).
2025-03-21T12:42:34.6695001Z make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
2025-03-21T12:42:34.6700052Z make[2]: Leaving directory '/home/runner/work/Phicomm-K3-LEDE-Firmware-Lean/Phicomm-K3-LEDE-Firmware-Lean/openwrt'
2025-03-21T12:42:34.6731992Z make[1]: *** [package/Makefile:110: /home/runner/work/Phicomm-K3-LEDE-Firmware-Lean/Phicomm-K3-LEDE-Firmware-Lean/openwrt/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
2025-03-21T12:42:34.6736843Z make[1]: Leaving directory '/home/runner/work/Phicomm-K3-LEDE-Firmware-Lean/Phicomm-K3-LEDE-Firmware-Lean/openwrt'
2025-03-21T12:42:34.6775550Z make: *** [/home/runner/work/Phicomm-K3-LEDE-Firmware-Lean/Phicomm-K3-LEDE-Firmware-Lean/openwrt/include/toplevel.mk:231: world] Error 2
2025-03-21T12:42:34.6787298Z ##[error]Process completed with exit code 2.

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

K3

### 详细日志

附件为完整的编译log
[14_编译固件 - Compile Firmware.txt](https://github.com/user-attachments/files/19389834/14_.-.Compile.Firmware.txt)

xzchsia, 2025-03-24:
经过测试，从3.14号(提交ID：3f938d01a6fd027e3e1d111ecc5061aba37e31c6  kernel: bump all to latest HEAD ) 开始的所有版本编译均失败。

coolsnowwolf, 2025-03-24:
已修复

closed 2025-03-25 by coolsnowwolf