coolsnowwolf/lede #13413: cmiot_ax18 启动时WAN有概率获取不到上级路由的DHCP ip

Game0fLife, 2025-03-22:
### 详细叙述

最新编译的固件，默认配置，接入wan口上电，第一次能正常获取ip，重启后，无法获取wan口ip，，再次重启能获取到

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

cmiot_ax18 

### 详细日志

日志见附件

[config.txt](https://github.com/user-attachments/files/19402351/config.txt)

[获取失败Sat Mar 22 17:49:18 2025 Kernel Log.txt](https://github.com/user-attachments/files/19402355/Sat.Mar.22.17.49.18.2025.Kernel.Log.txt)
[获取失败Sat Mar 22 17:49:18 2025 系统日志.txt](https://github.com/user-attachments/files/19402357/Sat.Mar.22.17.49.18.2025.txt)
[获取成功Sat Mar 22 17:55:43 2025 Kernel Log.txt](https://github.com/user-attachments/files/19402356/Sat.Mar.22.17.55.43.2025.Kernel.Log.txt)
[获取成功Sat Mar 22 17:55:43 2025 系统日志.txt](https://github.com/user-attachments/files/19402354/Sat.Mar.22.17.55.43.2025.txt)

Game0fLife, 2025-03-22:
make 日志：https://1drv.ms/t/c/6eefba6f36c1b68e/Ec8sH75lmR9PsDikbGeg03kBRrRtBvTqBKf-7fqxdQHMPw?e=evxJT1usb2、usb3 U盘也无法识别

1980490718, 2025-03-25:
切换老版luci应该可以
src-git packages https://github.com/coolsnowwolf/packages
src-git luci https://github.com/coolsnowwolf/luci
#src-git luci https://github.com/coolsnowwolf/luci.git;openwrt-23.05
最近的源码用最新的luci确实有usb挂载，硬盘挂载，以及smb的问题

Game0fLife, 2025-03-26:
> 更改feeds.conf.default 切换老版luci应该可以 src-git packages https://github.com/coolsnowwolf/packages src-git luci https://github.com/coolsnowwolf/luci #src-git luci https://github.com/coolsnowwolf/luci.git;openwrt-23.05 最近的源码用最新的luci确实有usb挂载，硬盘挂载，以及smb的问题 [@Game0fLife](https://github.com/Game0fLife)

感谢回复，我切换一下老版luci编译试试

Game0fLife, 2025-03-27:
使用老版luci编译后，设备关机重启几次试了下，wan口可以正常获取IP了，但usb挂载还是不能识别

closed 2025-03-27 by Game0fLife