coolsnowwolf/lede #13420: 最新编译的固件，dockerman无法访问容器，配置里没有 添加允许访问的接口 选项

c455025066, 2025-03-25:
### 详细叙述

详细看了两个版本的docker，版本号都一样，但是最新编译的dockerman，配置里面没有 [添加允许访问的接口。导致我启动容器以后无法进入容器的后台，但是旧版本的docker里面是有这个配置的，具体如图。这个是什么原因呢？求各位大佬帮帮忙](u

[image]

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

X86

### 详细日志

[image]

coolsnowwolf, 2025-03-27:
老版本的 dockerman luci 才有这个功能

c455025066, 2025-03-28:
> 老版本的 dockerman luci 才有这个功能

那大佬现在怎么设置才能进容器后台啊？

hong0980, 2025-04-07:
现在的版本好像只有uci set firewall.@defaults[0].forward='ACCEPT'才能进容器后台

coolsnowwolf, 2026-09-05:
当前仓库的 luci-app-dockerman 已采用新版 LuCI 页面结构，容器网络与端口映射在创建/编辑容器页面中配置，不再沿用旧版“添加允许访问的接口”字段；该字段属于旧 dockerman UI。原 issue 的 maintainer 后续已说明已增加相关选项，但没有当前版本复现信息，故关闭。

closed 2026-09-05 by coolsnowwolf