coolsnowwolf/lede #13421: Debian12报错 /home/juneover24/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnnyyynynyyyyyyynyyynynyynnynnyyyyyynyyyyyyyyyyynynnyyyyynyyy] Error 2

SolitaryJune, 2025-03-25:
### 详细叙述

跑不起来
make[1]: Leaving directory '/home/juneover24/lede'
make: *** [/home/juneover24/lede/include/toplevel.mk:231：world] 错误 2

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

x86 和 redmi ax6000

### 详细日志

ln -fs libdw-0.191.so /home/juneover24/lede/staging_dir/host/lib/libdw.so.1
ln -fs libdw.so.1 /home/juneover24/lede/staging_dir/host/lib/libdw.so
make[6]: Nothing to be done for 'install-exec-am'.
 /usr/bin/mkdir -p '/home/juneover24/lede/staging_dir/host/include/elfutils'
 /home/juneover24/lede/staging_dir/host/bin/install -c -m 644 version.h '/home/juneover24/lede/staging_dir/host/include/elfutils'
make[4]: Leaving directory '/home/juneover24/lede/build_dir/host/elfutils-0.191'
mkdir -p /home/juneover24/lede/staging_dir/host/stamp
touch /home/juneover24/lede/build_dir/host/elfutils-0.191/.built
touch /home/juneover24/lede/staging_dir/host/stamp/.elfutils_installed
make[3]: Leaving directory '/home/juneover24/lede/tools/elfutils'
time: tools/elfutils/compile#54.55#15.53#134.44
make[2]: Leaving directory '/home/juneover24/lede'
make[1]: *** [tools/Makefile:228: /home/juneover24/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnnyyynynyyyyyyynyyynynyynnynnyyyyyynyyyyyyyyyyynynnyyyyynyyy] Error 2
make[1]: Leaving directory '/home/juneover24/lede'
make: *** [/home/juneover24/lede/include/toplevel.mk:231：world] 错误 2
juneover24@debian:~/lede$

d975xbx, 2025-03-26:
网络问题。Debian12和ubuntu24.04在win和macos里都正常

SolitaryJune, 2025-03-26:
> 网络问题。Debian12和ubuntu24.04在win和macos里都正常

是WSL吗，我在hyper-v里面跑的Debian12

shjy10000, 2025-03-27:
网的的问题，切换节点，删除缓存重新编译就好了

coolsnowwolf, 2026-09-05:
日志只显示 `tools/elfutils` 的安装完成以及上层 `world` 汇总错误，没有真正失败的命令、退出原因或 WSL 环境信息。当前 LEDE 的 elfutils 主机工具已包含 macOS/现代编译器兼容修复并可成功构建；该 issue 无法据此定位 Debian 12/WSL 的源码问题，先关闭。若重现请提供第一次出现的 `Error 1` 前后完整日志。

closed 2026-09-05 by coolsnowwolf