coolsnowwolf/lede #13422: make menuconfig后没有nikki这个插件的选项

woni928, 2025-03-26:
### 详细叙述

执行
./scripts/feeds update -a
./scripts/feeds install -a
make defconfig
怎么会没有，姿势不对吗？哪位同学指点一下，谢谢！

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

N1

### 详细日志

[image]

d975xbx, 2025-03-26:
添加源

woni928, 2025-03-26:
> 添加源 具体看这个网址：https://github.com/nikkinikki-org/OpenWrt-nikki

源添加了的

chen-wilde, 2025-03-31:
需要在 Base system 启用 firewall4

woni928, 2025-03-31:
> 需要在 Base system 启用 firewall4

如何在云编译的时候启动呢？请教一下

chen-wilde, 2025-03-31:
> > 需要在 Base system 启用 firewall4
> 
> 如何在云编译的时候启动呢？请教一下

CONFIG_PACKAGE_firewall4=y

woni928, 2025-04-01:
> > > 需要在 Base system 启用 firewall4
> > 
> > 
> > 如何在云编译的时候启动呢？请教一下
> 
> CONFIG_PACKAGE_firewall4=y

感谢，测试了下  还是没有

mmeiwenlong, 2025-04-17:
> 哪位同学指点一下
需要23.05开始，，，

closed 2025-04-17 by woni928