coolsnowwolf/lede #13423: PHP8编译报错

wzb261643358, 2025-03-27:
### 详细叙述

2025-03-25T14:02:30.7268007Z configure: error: XMLReader extension requires LIBXML extension, add --with-libxml
2025-03-25T14:02:30.7814983Z make[3]: *** [Makefile:633: /home/runner/work/lede_action_openwrt/lede_action_openwrt/build_dir/target-x86_64_musl/php-8.2.2/.configured_db9dc9cca851eac5748573e77ca1d958] Error 1
2025-03-25T14:02:30.7816224Z make[3]: Leaving directory '/home/runner/work/lede_action_openwrt/lede_action_openwrt/feeds/packages/lang/php8'
2025-03-25T14:02:30.7819625Z time: package/feeds/packages/php8/compile#8.17#7.18#14.00
2025-03-25T14:02:30.7827330Z     ERROR: package/feeds/packages/php8 failed to build.

报错
https://github.com/wzb261643358/lede_action_openwrt
本人主页

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

x86-64

### 详细日志

2025-03-25T14:02:30.7268007Z configure: error: XMLReader extension requires LIBXML extension, add --with-libxml
2025-03-25T14:02:30.7814983Z make[3]: *** [Makefile:633: /home/runner/work/lede_action_openwrt/lede_action_openwrt/build_dir/target-x86_64_musl/php-8.2.2/.configured_db9dc9cca851eac5748573e77ca1d958] Error 1
2025-03-25T14:02:30.7816224Z make[3]: Leaving directory '/home/runner/work/lede_action_openwrt/lede_action_openwrt/feeds/packages/lang/php8'
2025-03-25T14:02:30.7819625Z time: package/feeds/packages/php8/compile#8.17#7.18#14.00
2025-03-25T14:02:30.7827330Z     ERROR: package/feeds/packages/php8 failed to build.

coolsnowwolf, 2025-03-27:
已修复

wzb261643358, 2025-03-27:
可以可以已经编译成功 ，但是我加入luci_app_docker 
为什么编译出来却没有这个app

closed 2025-04-22 by coolsnowwolf