coolsnowwolf/lede #13426: 兼容性破坏:RTL8367B

zfdx123, 2025-03-28:
### 详细叙述

问题在 https://github.com/coolsnowwolf/lede/issues/13393 中被提出，在跟踪提交发现在 https://github.com/coolsnowwolf/lede/commit/05390b546e39f8b4caf7edc573440dcc42b44b4b#diff-191b4f6529da5644892e6206c09014e70afa001fbc2f845a92da20574cb28252L748 中删除了之前的兼容性代码该代码提交为 https://github.com/coolsnowwolf/lede/commit/985fa051b03b0fd7dbe38268dd5f73933f671a91 目前待确认兼容性是否是完全被破坏，等待 问题13393回复

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

RTL8367B 交换机驱动

### 详细日志

无

aiamadeus, 2025-03-28:
您这帽子扣得牛逼啊，别是瞎猜的吧？
chip_ver == 0x0020 直接 return 这 j8 代码
纯粹是因为不知道寄存器怎么配避免灯不亮
怎么到你这就变成我们在破坏兼容性了
我怀疑你压根就没有测试过

zfdx123, 2025-03-28:
是的，确实是猜测，在此对我的言辞表达表示抱歉，但是在前边13393的开启问题中，并没有引起重视，上查commit只有这次提交可能有点关系，再次表达歉意，但是希望可以解决相关用户问题，因为该代码当时确实解决了相关设备，某一网口无法启动的问题

aiamadeus, 2025-03-30:
https://github.com/coolsnowwolf/lede/commit/9914389a8ddc2966f5b27672165e5c98f8ed903a 我想这个已经修了
没有日志的情况下最好不要瞎猜

zfdx123, 2025-03-30:
https://github.com/coolsnowwolf/lede/issues/13393#issuecomment-2762962093

他似乎编译了最新版本，依然不可用，不好意思，我对瞎猜抱歉

zfdx123, 2025-03-30:
等待一下他的最新版本编译和日志

zfdx123, 2025-05-04:
[image]

错误未复现，交换机状态正常，所有端口可用

closed 2025-05-04 by zfdx123