coolsnowwolf/lede #13429: macos m4原生编译问题

metaforget, 2025-03-30:
### 详细叙述

最近购入mac mini m4尝试原生编译，使用了readme中推荐的brew包及环境变量设置，但是一直不行。
换用官方op仓库，直接编译成功，并且gcc使用的是苹果clang

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

mac mini m4

### 详细日志

```bash
PATH="/opt/homebrew/opt/make/libexec/gnubin:$PATH"
PATH="/opt/homebrew/opt/gnu-getopt/bin:$PATH"
PATH="/opt/homebrew/opt/gettext/bin:$PATH"
PATH="/opt/homebrew/opt/coreutils/libexec/gnubin:$PATH"
PATH="/opt/homebrew/opt/findutils/libexec/gnubin:$PATH"
PATH="/opt/homebrew/opt/gnu-tar/libexec/gnubin:$PATH"
PATH="/opt/homebrew/opt/grep/libexec/gnubin:$PATH"
PATH="/opt/homebrew/opt/gnu-sed/libexec/gnubin:$PATH"
PATH="/opt/homebrew/opt/gcc/bin:$PATH"
PATH="/opt/homebrew/bin:$PATH"
export PATH

```

但是直接报错，这个elfutils

```bash
checking whether gcc supports __attribute__((visibility()))... yes
checking whether gcc supports __attribute__((gcc_struct))... no
checking whether gcc supports -fPIC... yes
checking whether gcc supports -fPIE... yes
checking whether gcc supports -Wl,-z,defs... no
checking whether the compiler generates build-ids... ./configure: line 6978: -n: command not found
no
configure: WARNING: compiler doesn't generate build-id by default
checking whether gcc supports -Wl,-z,relro... no
checking for __thread support... no
configure: error: __thread support required
make[3]: *** [Makefile:38: /Volumes/OpenWrt/lede/build_dir/host/elfutils-0.191/.configured] Error 1
make[3]: Leaving directory '/Volumes/OpenWrt/lede/tools/elfutils'
time: tools/elfutils/compile#3.31#0.77#5.58
    ERROR: tools/elfutils failed to build.
make[2]: *** [tools/Makefile:232: tools/elfutils/compile] Error 1
make[2]: Leaving directory '/Volumes/OpenWrt/lede'
make[1]: *** [tools/Makefile:228: /Volumes/OpenWrt/lede/staging_dir/host/stamp/.tools_compile_nyyynyyynnnyyyyynyyyyyyynyyynynyynnynnyyyyyynyyyyyyyyyyynynnyyyyynyyy] Error 2
make[1]: Leaving directory '/Volumes/OpenWrt/lede'
make: *** [/Volumes/OpenWrt/lede/include/toplevel.mk:231: world] Error 2
```

找大模型看了一下，发现staging_dir/host/bin/gcc指向的是主机苹果clang，又加了下面变量

```bash
export CC=gcc-11  
export CXX=g++-11
```

make dirclean清除后重新编译，又提示下面报错

```bash
subprocess.CalledProcessError: Command '/Volumes/OpenWrt/lede/staging_dir/host/bin/g++ -MMD -MT build/browse.o -MF build/browse.o.d -g -Wall -Wextra -Wno-deprecated -Wno-missing-field-initializers -Wno-unused-parameter -fno-rtti -fno-exceptions -std=c++11 -fvisibility=hidden -pipe '-DNINJA_PYTHON="python3"' -O2 -DNDEBUG -fdiagnostics-color -DNINJA_HAVE_BROWSE -I. -O2 -I/Volumes/OpenWrt/lede/staging_dir/host/include  -O2 -I/Volumes/OpenWrt/lede/staging_dir/host/include  -c ./src/browse.cc -o build/browse.o' returned non-zero exit status 2.
make[3]: *** [Makefile:37: /Volumes/OpenWrt/lede/build_dir/host/ninja-1.12.1/.configured] Error 1
make[3]: Leaving directory '/Volumes/OpenWrt/lede/tools/ninja'
time: tools/ninja/compile#0.13#0.17#0.43
    ERROR: tools/ninja failed to build.
make[2]: *** [tools/Makefile:232: tools/ninja/compile] Error 1
make[2]: Leaving directory '/Volumes/OpenWrt/lede'
make[1]: *** [tools/Makefile:228: /Volumes/OpenWrt/lede/staging_dir/host/stamp/.tools_compile_nyyynyyynnnyyyyynyyyyyyynyyynynyynnynnyyyyyynyyyyyyyyyyynynnyyyyynyyy] Error 2
make[1]: Leaving directory '/Volumes/OpenWrt/lede'
make: *** [/Volumes/OpenWrt/lede/include/toplevel.mk:231: world] Error 2
```

metaforget, 2025-03-30:
实际是有类似issue，但是都没说解决

coolsnowwolf, 2026-09-05:
已加入 macOS/Apple Clang 的 elfutils common-symbol 修复补丁，提交并推送：`68a496c92ab`。补丁将 `__libelf_fill_byte` 显式初始化，避免 macOS 默认 `-fno-common` 链接失败；Linux 主机 `make tools/elfutils/compile -j30 V=s` 已通过。

closed 2026-09-05 by coolsnowwolf