coolsnowwolf/lede #13432: 6.12 x86编译失败

whatpig, 2025-04-01:
### 详细叙述

6.6同样的config没有问题

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

x86

### 详细日志

Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.12.20-1-81d13be5a8fba7e2a1588100e732da52) for kmod-button-hotplug
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-button-hotplug found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-button-hotplug.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.12.20-1-81d13be5a8fba7e2a1588100e732da52) for kmod-cryptodev
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-cryptodev found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-cryptodev.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.12.20-1-81d13be5a8fba7e2a1588100e732da52) for kmod-r8125-rss
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-r8125-rss found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-r8125-rss.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.12.20-1-81d13be5a8fba7e2a1588100e732da52) for kmod-r8126-rss
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-r8126-rss found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-r8126-rss.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.12.20-1-81d13be5a8fba7e2a1588100e732da52) for kmod-r8168
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-r8168 found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-r8168.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.12.20-1-81d13be5a8fba7e2a1588100e732da52) for kmod-usb-net-rtl8152-vendor
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-usb-net-rtl8152-vendor found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-usb-net-rtl8152-vendor.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.12.20-1-81d13be5a8fba7e2a1588100e732da52) for kmod-ipt-fullconenat
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-fullconenat found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-ipt-fullconenat.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ddns-scripts_aliyun:
 * 	kernel (= 6.12.20-1-81d13be5a8fba7e2a1588100e732da52)
 * opkg_install_cmd: Cannot install package ddns-scripts_aliyun.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for firewall:
 * 	kernel (= 6.12.20-1-81d13be5a8fba7e2a1588100e732da52)
 * opkg_install_cmd: Cannot install package firewall.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/x86s/lede'
make[1]: *** [package/Makefile:111: /home/x86s/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/x86s/lede'
make: *** [/home/x86s/lede/include/toplevel.mk:231：world] 错误 2

loydcheung, 2025-04-01:
kernel 6.6，同样的问题

mumuou, 2025-04-01:
就最新那个更新的锅，乱删除依赖，不测试就合并。

bi4tms, 2025-04-01:
删除编译目录下的build-dir->target-x86_64-musi->linux-x86_64文件夹，然后重新编译即可

bi4tms, 2025-04-01:
> 就最新那个更新的锅，乱删除依赖，不测试就合并。
谦虚一点，多搜一搜issue

closed 2025-04-11 by coolsnowwolf