coolsnowwolf/lede #13437: Issues Activating Wi-Fi 6 (802.11ax) and Bluetooth on Sige7

dalnaraD, 2025-04-02:
### 详细叙述

Sige7 is running on compiled iStoreOS22 and 24 firmware.
LuCI only enables 802.11bg, and 802.11ax (Wi-Fi 6) activation is unsuccessful.
Despite compiling and running hostapd, even 802.11ac fails to load properly.
BlueZ was compiled, but the Bluetooth driver does not activate, possibly due to unsupported drivers.
Questions
I am using a 5V 2A USB to C adapter. Is it related to the power?
How to enable 802.11ax?
How to resolve hostapd issues?
Any solutions to activate the Bluetooth driver?

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

Sige7 (BPI-M7)

### 详细日志

Firmware Logs [ 616.974177] [dhd-pcie] dhd_bus_download_firmware: firmware path=/lib/firmware/rkwifi/fw_bcmdhd.bin, nvram path=/lib/firmware/rkwifi/nvram.txt [ 616.974247] [dhd-pcie] dhd_conf_set_path_params : Final fw_path=/lib/firmware/rkwifi/fw_bcm43752a2_pcie_ag.bin [ 616.974257] [dhd-pcie] dhd_conf_set_path_params : Final nv_path=/lib/firmware/rkwifi/nvram_AP6275P.txt [ 616.974267] [dhd-pcie] dhd_conf_set_path_params : Final clm_path=/lib/firmware/rkwifi/clm_bcm43752a2_pcie_ag.blob [ 616.974277] [dhd-pcie] dhd_conf_set_path_params : Final conf_path=/lib/firmware/rkwifi/config_bcm43752a2_pcie_ag.txt [ 616.974334] [dhd-pcie] dhd_os_open_image1: /lib/firmware/rkwifi/config_bcm43752a2_pcie_ag.txt (336 bytes) open success [ 616.974725] [dhd-pcie] dhdpcie_download_code_file: download firmware /lib/firmware/rkwifi/fw_bcm43752a2_pcie_ag.bin [ 616.974765] [dhd-pcie] dhd_os_open_image1: /lib/firmware/rkwifi/fw_bcm43752a2_pcie_ag.bin (936074 bytes) open success [ 617.070192] [dhd-pcie] dhd_os_open_image1: /lib/firmware/rkwifi/nvram_AP6275P.txt (7458 bytes) open success [ 617.198072] [dhd-pcie] dhd_legacy_preinit_ioctls: use firmware generated mac_address 54:78:c9:29:4b:52 [ 617.198093] [dhd-pcie] dhd_os_open_image1: /lib/firmware/rkwifi/clm_bcm43752a2_pcie_ag.blob (30993 bytes) open success [ 617.257775] [dhd-pcie] Firmware up: op_mode=0x0005, MAC=54:78:c9:29:4b:52 [ 617.304515] [dhd-pcie] Firmware: wl0: Jul 12 2022 18:54:54 version 18.35.387.23.146 (g412cc5ec) FWID 01-93c53be6

dalnaraD, 2025-04-02:
> [image] ??

Sorry for the confusion.

The latest OpenWrt supports the BPI-M7, but WiFi does not work.
Why is that?

If I port and load the AP6275P (BCM43752) driver, will it work with OpenWrt LuCI?

coolsnowwolf, 2026-09-05:
日志显示的是 Broadcom 厂商 `dhd-pcie` 驱动和 AP6275P 专用固件已经加载（`fw_bcm43752a2_pcie_ag.bin`、`nvram_AP6275P.txt`），并非 hostapd 编译失败；Wi-Fi 6 能力取决于该厂商驱动/固件的能力声明。当前 LEDE 的 Sige7 镜像使用主线 `kmod-brcmfmac`，没有可直接启用该 dhd-pcie/AP6275P 驱动的源码与固件包；Bluetooth 也需要对应 UART/PCIe 控制器驱动和固件。该第三方驱动/固件移植超出当前 issue 提供的信息，暂无安全的 LEDE 源码修复，先关闭。

closed 2026-09-05 by coolsnowwolf