coolsnowwolf/lede #13441: 麻烦修复一下MT7916 开启160m的握手速率bug

cooppp, 2025-04-08:
### 详细叙述

在这个分支中https://github.com/DHDAXCW/lede-rockchip，mt7916的160m可以正常运行

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

rk3568-opc-h69k

### 详细日志

[image]
[image]

cooppp, 2025-04-08:
找到，mt76: fix mt7916 160Mhz support in kernel 5.4：[Commit 97d3d1c](https://github.com/coolsnowwolf/lede/commit/97d3d1c6aaf0e3a9c63a58316234b5221081ad1b)，但是编译出现以下报错。

make[4]: Entering directory '/home/sc/lede/target/linux/rockchip'
rm -f /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/localversion
/home/sc/lede/scripts/kconfig.pl  + /home/sc/lede/target/linux/generic/config-5.4 /home/sc/lede/target/linux/rockchip/armv8/config-5.4 > /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/sc/lede/.config >> /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.target
/home/sc/lede/scripts/package-metadata.pl kconfig /home/sc/lede/tmp/.packageinfo /home/sc/lede/.config 5.4 > /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.override
/home/sc/lede/scripts/kconfig.pl 'm+' '+' /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.target /dev/null /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.override > /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.set
mv /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.set /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.old
grep -v INITRAMFS /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.old > /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.set
echo "# CONFIG_INITRAMFS_PRESERVE_MTIME is not set" >> /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.set
rm -rf /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/modules
cmp -s /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.set /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.prev || { cp /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.set /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config; cp /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.set /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.prev; }
export MAKEFLAGS= ; [ -d /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/user_headers ] || make -C /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291 KCFLAGS="-fmacro-prefix-map=/home/sc/lede/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/sc/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Apr  7 13:00:02 2025" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/sc/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.291 CC="aarch64-openwrt-linux-musl-gcc"  INSTALL_HDR_PATH=/home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/user_headers headers_install
grep '=[ym]' /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.config.set | LC_ALL=C sort | /home/sc/lede/staging_dir/host/bin/mkhash md5 > /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.vermagic
touch /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.configured
rm -f /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/vmlinux /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/System.map
make -C /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291 KCFLAGS="-fmacro-prefix-map=/home/sc/lede/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/sc/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Apr  7 13:00:02 2025" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/sc/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRE [... cut ...]
make[4]: *** [Makefile:26: /home/sc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.291/.modules] Error 2
make[4]: Leaving directory '/home/sc/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/sc/lede/target/linux'
time: target/linux/compile#5.09#3.17#8.23
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/sc/lede'
make[1]: *** [target/Makefile:23: /home/sc/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2

cooppp, 2025-04-10:
> 你用5.4就是个错误

主要还是160m在默认6.6.68内核中是无法正常使用的，看到支持[Commit 97d3d1c](https://github.com/coolsnowwolf/lede/commit/97d3d1c6aaf0e3a9c63a58316234b5221081ad1b)
5.4中修复160m支持才用的

coolsnowwolf, 2025-04-12:
直接编译新版本即可

aiamadeus, 2025-04-27:
[image]
有没有可能 rk3568 和其他路由器上**板载**的 mt7916 很正常，只是你们的 asiarf 网卡 efuse 有问题而已
不可能为了少数人在用的网卡而去破坏别人在正常使用的网卡（比如 #13448 直接屏蔽 6G）。。。

closed 2025-05-02 by cooppp