coolsnowwolf/lede #13443: x86下phy/b53编译出错

ilxp, 2025-04-11:
### 详细叙述

phy/b53编译出错。
日志如下：

各位大佬，怎么解

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

x64

### 详细日志

make[5]: Entering directory '/builder/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.22'
  DESCEND objtool
  INSTALL libsubcmd_headers
  CC [M]  drivers/net/phy/b53/b53_common.o
drivers/net/phy/b53/b53_common.c:230:6: error: no previous prototype for 'b53_set_forwarding' [-Werror=missing-prototypes]
  230 | void b[53](https://github.com/ilxp/oprx-builder/actions/runs/14401760679/job/40388913621#step:8:54)_set_forwarding(struct b53_device *dev, int enable)
      |      ^~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[11]: *** [scripts/Makefile.build:229: drivers/net/phy/b53/b53_common.o] Error 1
make[10]: *** [scripts/Makefile.build:478: drivers/net/phy/b53] Error 2
make[9]: *** [scripts/Makefile.build:478: drivers/net/phy] Error 2
make[8]: *** [scripts/Makefile.build:478: drivers/net] Error 2
make[7]: *** [scripts/Makefile.build:478: drivers] Error 2
make[6]: *** [/builder/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.22/Makefile:1944: .] Error 2
make[5]: *** [Makefile:224: __sub-make] Error 2
make[5]: Leaving directory '/builder/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.22'
make[4]: *** [Makefile:28: /builder/openwrt/build_dir/target-x86_[64](https://github.com/ilxp/oprx-builder/actions/runs/14401760679/job/40388913621#step:8:65)_musl/linux-x86_64/linux-6.12.22/.modules] Error 2
make[4]: Leaving directory '/builder/openwrt/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/builder/openwrt/target/linux'
time: target/linux/compile#2.08#0.92#2.94
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/builder/openwrt'
make[1]: *** [target/Makefile:23: /builder/openwrt/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/builder/openwrt'
make: *** [/builder/openwrt/include/toplevel.mk:231: world] Error 2

coolsnowwolf, 2025-04-11:
这个bcm 的交换机芯片，哪个 x86 主板集成了？？？

ilxp, 2025-04-11:
我也是看了，但是编译就是报错。看来只有删除乐

ilxp, 2025-06-05:
不知道怎么回事，在6.12内核下一直出现这个b53错误，在6.6以及以下都没有出现过：

# No change to .config
#
make[5]: Leaving directory '/builder/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.31'
make -C /builder/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.31 KCFLAGS="-fmacro-prefix-map=/builder/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/builder/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="OprX" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Wed Jun  4 04:27:05 2025" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/builder/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=6.12.31 CC="x86_64-openwrt-linux-musl-gcc" bzImage modules
make[5]: Entering directory '/builder/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.31'
  DESCEND objtool
  INSTALL libsubcmd_headers
Warning: Kernel ABI header at 'tools/arch/x86/include/asm/orc_types.h' differs from latest version at 'arch/x86/include/asm/orc_types.h'
diff -u tools/arch/x86/include/asm/orc_types.h arch/x86/include/asm/orc_types.h
Warning: Kernel ABI header at 'tools/arch/x86/include/asm/insn.h' differs from latest version at 'arch/x86/include/asm/insn.h'
diff -u tools/arch/x86/include/asm/insn.h arch/x86/include/asm/insn.h
Warning: Kernel ABI header at 'tools/arch/x86/lib/insn.c' differs from latest version at 'arch/x86/lib/insn.c'
diff -u tools/arch/x86/lib/insn.c arch/x86/lib/insn.c
  CC [M]  drivers/net/phy/b53/b53_mdio.o
drivers/net/phy/b53/b53_mdio.c:403:12: error: no previous prototype for 'b53_phy_driver_register' [-Werror=missing-prototypes]
  403 | int __init b53_phy_driver_register(void)
      |            ^~~~~~~~~~~~~~~~~~~~~~~
drivers/net/phy/b53/b53_mdio.c:425:13: error: no previous prototype for 'b53_phy_driver_unregister' [-Werror=missing-prototypes]
  425 | void __exit b53_phy_driver_unregister(void)
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[11]: *** [scripts/Makefile.build:229: drivers/net/phy/b53/b53_mdio.o] Error 1
make[10]: *** [scripts/Makefile.build:478: drivers/net/phy/b53] Error 2
make[9]: *** [scripts/Makefile.build:478: drivers/net/phy] Error 2
make[8]: *** [scripts/Makefile.build:478: drivers/net] Error 2
make[7]: *** [scripts/Makefile.build:478: drivers] Error 2
make[6]: *** [/builder/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.31/Makefile:1945: .] Error 2
make[5]: *** [Makefile:224: __sub-make] Error 2
make[5]: Leaving directory '/builder/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.31'
make[4]: *** [Makefile:28: /builder/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.31/.modules] Error 2
make[4]: Leaving directory '/builder/openwrt/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/builder/openwrt/target/linux'
time: target/linux/compile#1.93#0.89#2.76
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/builder/openwrt'
make[1]: *** [target/Makefile:23: /builder/openwrt/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/builder/openwrt'
make: *** [/builder/openwrt/include/toplevel.mk:231: world] Error 2

经过查找资料，看前面大佬说是，

closed 2025-06-05 by ilxp