coolsnowwolf/lede #13444: 6.12.23编译错误

Jason6111, 2025-04-12:
### 详细叙述

编译错误

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

x86

### 详细日志

Applying /mnt/workdir/openwrt/target/linux/generic/hack-6.12/220-arm-gc_sections.patch using plaintext: 
patching file arch/arm/Kconfig
patching file arch/arm/boot/compressed/Makefile
patching file arch/arm/kernel/vmlinux.lds.S
patching file arch/arm/include/asm/vmlinux.lds.h
Hunk #1 succeeded at 54 (offset 6 lines).
Hunk #2 succeeded at 114 (offset 6 lines).
Hunk #3 FAILED at 125.
Hunk #4 succeeded at 149 (offset 6 lines).
1 out of 4 hunks FAILED -- saving rejects to file arch/arm/include/asm/vmlinux.lds.h.rej
Patch failed!  Please fix /mnt/workdir/openwrt/target/linux/generic/hack-6.12/220-arm-gc_sections.patch!
make[4]: *** [Makefile:35: /mnt/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.23/.prepared_4bf367f4815076a492c23872ded064ff] Error 1
make[4]: Leaving directory '/mnt/workdir/openwrt/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/mnt/workdir/openwrt/target/linux'
time: target/linux/compile#8.44#7.75#11.82
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/mnt/workdir/openwrt'
make[1]: *** [target/Makefile:23: /mnt/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/mnt/workdir/openwrt'
make: *** [/mnt/workdir/openwrt/include/toplevel.mk:231: world] Error 2

GreatMichaelLee, 2025-04-12:
+1 同样错误，patch有问题，麻烦老大修复下。

ilxp, 2025-04-12:
删除即可 target/linux/generic/hack-6.12/220-arm-gc_sections.patch

closed 2025-04-12 by Jason6111