coolsnowwolf/lede #13455: package/feeds/packages/ruby/Makefile  yaml/host

coolveyron, 2025-04-27:
### 详细叙述

开了helloworld，make menuconfig之后就提示：
WARNING: Makefile 'package/feeds/packages/ruby/Makefile' has a build dependency on 'yaml/host', which does not exist

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

X86

### 详细日志

Installing package 'lua-neturl' from helloworld
Installing package 'luci-app-ssr-plus' from helloworld
Installing package 'shadowsocksr-libev' from helloworld
Installing package 'v2ray-core' from helloworld
Installing package 'tuic-client' from helloworld
Installing package 'shadow-tls' from helloworld
Installing package 'naiveproxy' from helloworld
Installing package 'redsocks2' from helloworld
Installing package 'shadowsocks-rust' from helloworld
Installing package 'simple-obfs' from helloworld
Installing package 'v2ray-plugin' from helloworld
Installing package 'xray-plugin' from helloworld
ecs-user@iZrj9923bneh2p0bk8lcwcZ:~/lede$ make menuconfig
Collecting package info: done
WARNING: Makefile 'package/feeds/packages/ruby/Makefile' has a build dependency on 'yaml/host', which does not exist

Your configuration changes were NOT saved.

ecs-user@iZrj9923bneh2p0bk8lcwcZ:~/lede$ 
ecs-user@iZrj9923bneh2p0bk8lcwcZ:~/lede$

cyrixiii, 2025-04-27:
我也遇到同样的问题

> ubuntu@ubuntu:~/lede$ ./scripts/feeds install -a
> WARNING: Makefile 'package/feeds/packages/ruby/Makefile' has a build dependency on 'yaml/host', which does not exist
> Installing all packages from feed packages.
> Installing all packages from feed luci.
> Installing all packages from feed routing.
> Installing all packages from feed telephony.
> Installing all packages from feed helloworld.

MilesPoupart, 2025-04-27:
ref: https://github.com/openwrt/packages/commit/a951e16646f284bfadf6b458d30ba069eb7db006

liuyanantj, 2025-04-27:
试了楼上的改法  还是编译失败啊

MilesPoupart, 2025-04-27:
> 试了楼上的改法 还是编译失败啊

https://github.com/openwrt/packages/blob/master/libs/yaml/Makefile
直接替换MakeFile试试？

liuyanantj, 2025-04-27:
> > 试了楼上的改法 还是编译失败啊
> 
> https://github.com/openwrt/packages/blob/master/libs/yaml/Makefile 直接替换MakeFile试试？

ist rust-std-1.85.1-x86_64-unknown-linux-musl
	finished in 3.714 seconds
Dist rust-std-1.85.1-x86_64-unknown-linux-gnu
	finished in 3.436 seconds
Dist rust-src-1.85.1
	finished in 0.629 seconds
WARNING: The `change-id` is missing in the `config.toml`. This means that you will not be able to track the major changes made to the bootstrap configurations.
NOTE: to silence this warning, add `change-id = 134650` at the top of `config.toml`
NOTE: this message was printed twice to make it more likely to be seen
Build completed successfully in 0:06:36
touch /home/tc/lede/build_dir/target-x86_64_musl/host/rustc-1.85.1-src/.built
( cd /home/tc/lede/build_dir/target-x86_64_musl/host/rustc-1.85.1-src/build/dist ; for targz in *.tar.gz; do /home/tc/lede/staging_dir/host/bin/libdeflate-gzip -dc "$targz" | tar -xf - ; done ; find . -mindepth 2 -maxdepth 2 -type f -name install.sh -execdir bash '{}' --prefix=/home/tc/lede/staging_dir/target-x86_64_musl/host --disable-ldconfig \; ; )
install: creating uninstall script at /home/tc/lede/staging_dir/target-x86_64_musl/host/lib/rustlib/uninstall.sh
install: installing component 'build-manifest'

    build manifest installed.

install: creating uninstall script at /home/tc/lede/staging_dir/target-x86_64_musl/host/lib/rustlib/uninstall.sh
install: installing component 'rust-src'

    rust src installed.

install: creating uninstall script at /home/tc/lede/staging_dir/target-x86_64_musl/host/lib/rustlib/uninstall.sh
install: installing component 'rust-std-x86_64-unknown-linux-musl'

    rust std installed.

install: creating uninstall script at /home/tc/lede/staging_dir/target-x86_64_musl/host/lib/rustlib/uninstall.sh
install: installing component 'cargo'

    cargo installed.

install: creating uninstall script at /home/tc/lede/staging_dir/target-x86_64_musl/host/lib/rustlib/uninstall.sh
install: installing component 'rustc'

    rustc installed.

install: creating uninstall script at /home/tc/lede/staging_dir/target-x86_64_musl/host/lib/rustlib/uninstall.sh
install: installing component 'rust-std-x86_64-unknown-linux-gnu'

    rust std installed.

mkdir -p /home/tc/lede/staging_dir/hostpkg/stamp
touch /home/tc/lede/build_dir/target-x86_64_musl/host/rustc-1.85.1-src/.built
touch /home/tc/lede/staging_dir/hostpkg/stamp/.rust_installed
make[3]: Leaving directory '/home/tc/lede/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#2955.37#840.96#450.19
make[2]: Leaving directory '/home/tc/lede'
make[1]: *** [package/Makefile:110: /home/tc/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/tc/lede'
make: *** [/home/tc/lede/include/toplevel.mk:231：world] 错误 2

liuyanantj, 2025-04-29:
我把整个lede文件夹删除了，然后重新搞了一遍，就好了  不知道为啥

15767905786, 2025-04-29:
我已经删除lede好几次重新构建，依旧有这个问题，妈的

closed 2025-04-30 by coolsnowwolf