coolsnowwolf/lede #13457: 最新编译的r2s固件无法启动, LAN口没反应

loveritsu929re, 2025-04-29:
### 详细叙述

使用ubuntu容器编译, 克隆本仓库为r2s单独的构建目录,编译过程正常, tf卡启动sys红灯闪烁之后常亮, LAN网口指示灯和LAN指示灯无任何反应, 接上也不下发IP地址; 网线改插WAN口,网口灯和指示灯都亮, 但是也没什么用

配置仅修改target为r2s, 无任何其他改动

openwrt.ai下载的镜像启动正常, 也排除了tf卡, 电源, 烧录软件的问题

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

NanoPi R2S

### 详细日志

编译正常

iHub-2020, 2025-05-01:
我也遇到了同样的情况，三个月初编译的时候还是好的

zuozhehao, 2025-05-07:
我也同样的情况，今天刚编译的

stonegr, 2025-05-08:
一样的情况

stonegr, 2025-05-08:
[image]
加上这个解决

stonegr, 2025-05-08:
> [image] 加上这个解决

再加上usb3

iHub-2020, 2025-05-08:
@stonegr 感谢分享，终于好了，谢谢！！

[image]

stonegr, 2025-05-09:
> [@stonegr](https://github.com/stonegr) 你这2个截图都一样，是加上哪几个。

图片上的3个再加上usb3

zuozhehao, 2025-05-12:
没有usb3 也可以正常启动

closed 2025-05-26 by coolsnowwolf