coolsnowwolf/lede #13460: 缺少组件

gssiwmd, 2025-04-30:
### 详细叙述

Already up to date.
Create index file './feeds/gssiwmd.index' 
Create index file './feeds/privatepackages.index' 
Collecting package info: done
Create index file './feeds/packages.index' 
Create index file './feeds/luci.index' 
Create index file './feeds/routing.index' 
Create index file './feeds/telephony.index' 
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-video-fwnode', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-video-async', which does not exist
Installing all packages from feed gssiwmd.
Installing all packages from feed privatepackages.
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-video-fwnode', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-video-async', which does not exist
#

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

无

### 详细日志

无

zhaochina2025, 2025-05-01:
我也是提示这个

coolveyron, 2025-05-01:
问题在lede/package/kernel/linux/modules/video.mk里面，我用官方的文件替换了这个，又提示缺少kmod-media-core和kmod-multimedia-input，我大概看了看，应该是与驱动有关系，官方的与L大的相互缺少，意思就是官方的有kmod-video-fwnode和kmod-video-async，L大的没有kmod-media-core和kmod-multimedia-input，不太敢动这个文件，等L大修复吧。

closed 2025-05-06 by coolsnowwolf