coolsnowwolf/lede #13463: design  主题不能使用

lpflhh, 2025-05-03:
### 详细叙述

最新固件，使用代码内置的design主题无法正常工作，在主页卡死

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

x86和其他所有型号

### 详细日志

最新固件，使用代码内置的design主题无法正常工作，在主页卡死

YoungsunLi, 2025-05-06:
俺也一样

Maguyusi, 2025-05-10:
> ### 详细叙述
> 最新固件，使用代码内置的design主题无法正常工作，在主页卡死
> 
> ### 重复 issue
> * [x]  没有类似的 issue
> 
> ### 具体型号
> x86和其他所有型号
> 
> ### 详细日志
> 最新固件，使用代码内置的design主题无法正常工作，在主页卡死

不只这一个主题不能使用 除了默认bootstrap和argon之外的都不能使用 跟luci的lua版本有关，
cd lede
vim feeds.conf.default
不要用23.05就好了

Maguyusi, 2025-05-14:
Wed May 14 10:34:28 2025 daemon.err uhttpd[2284]

[image]

liuyanantj, 2025-05-20:
我的也是一样 进不去管理页面  最后命令行给重置了才行

shjy223534, 2025-05-25:
是的，这个问题不知道啥时候开始的，一开始还能用的，后面就直接卡死

jw10126121, 2025-09-09:
这不是一直都不能用的吗？没有一次正常过

coolsnowwolf, 2026-09-05:
当前 feeds/luci 的 `luci-theme-design` 已持续更新到 2026/06/16（commit 88bb18203d），并已切换到 25.12 ucode 结构及后续兼容修复；原 issue 没有提供当前版本的完整 uhttpd/LuCI 堆栈，无法确认仍存在同一问题。请先更新 feeds/luci 后测试，现关闭。

closed 2026-09-05 by coolsnowwolf