coolsnowwolf/lede #13467: 添加插件strongswan后，打不开。

zhaochina2025, 2025-05-05:
### 详细叙述

RPCError
RPC call to uci/get failed with ubus code 4: 未找到资源
  at ClassConstructor.handleCallReply (http://192.168.1.1/luci-static/resources/rpc.js?v=git-25.110.68714-844a8ee:94:7)

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

JDcloud ER1

### 详细日志

刷机后的提示
RPCError
RPC call to uci/get failed with ubus code 4: 未找到资源
  at ClassConstructor.handleCallReply (http://192.168.1.1/luci-static/resources/rpc.js?v=git-25.110.68714-844a8ee:94:7)

zhaochina2025, 2025-05-08:
我已经编译成功并安装到jdcloud er1路由器上，在编译strongswan插件时，配置选项选择了strongswan-default，就这样在网页管理页面打不开。

liangyuan64, 2025-05-21:
> 我已经编译成功并安装到jdcloud er1路由器上，在编译strongswan插件时，配置选项选择了strongswan-default，就这样在网页管理页面打不开。

touch /etc/config/ipsec

zhaochina2025, 2025-05-22:
> touch /etc/config/ipsec

thanks!

zhaochina2025, 2025-05-22:
> touch /etc/config/ipsec

luci-app-libreswan 安装后也是这个提示，touch /etc/config/ipsec    还是报错。

liangyuan64, 2025-05-27:
> > touch /etc/config/ipsec
> 
> luci-app-libreswan 安装后也是这个提示，touch /etc/config/ipsec 还是报错。

试试 ubus monitor 命令  点 web目录错误页面，debug一下。可以看到缺什么

coolsnowwolf, 2026-09-05:
已核对当前 strongSwan 打包：`strongswan-default` 依赖 `strongswan-swanctl`，后者安装 `/etc/config/ipsec` 和迁移脚本；LuCI RPC 查询该 UCI 配置不应再因文件不存在而失败。原 issue 使用旧版本且后续反馈仅建议手动 touch，当前包结构已包含正式配置文件，未发现仍需提交的源码修复，故关闭。

closed 2026-09-05 by coolsnowwolf