coolsnowwolf/lede #13469: ERROR: tools/firmware-utils failed to build.

jackzzg, 2025-05-08:
### 详细叙述

下载报错：
ERROR: tools/firmware-utils failed to build.

make 下载链接，提示文件不存在。
https://sources.openwrt.org/firmware-utils-2024-10-20-4b763892.tar.xz

仓库地址，文件名称不一致。 仓库的包名称是：
https://sources.openwrt.org/firmware-utils-2024.10.20~4b763892.tar.zst

2024.10.20~4b763892.tar.zst
2024-10-20-4b763892.tar.xz

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

Ubuntu

### 详细日志

    ERROR: tools/firmware-utils failed to build.
    ERROR: package/network/utils/iwinfo failed to build.
    ERROR: package/libs/libubox failed to build.
    ERROR: package/system/ubus failed to build.
    ERROR: package/system/rpcd failed to build.

jackzzg, 2025-05-08:
cd dl

手动下载：
wget https://sources.openwrt.org/libiwinfo-2025.02.06~9cec6b4d.tar.zst
wget https://sources.openwrt.org/ubus-2024.10.20~252a9b0c.tar.zst
wget https://sources.openwrt.org/rpcd-2024.12.02~cc9a471c.tar.zst
wget https://sources.openwrt.org/libubox-2024.03.29~eb9bcb64.tar.xz
wget https://sources.openwrt.org/firmware-utils-2024.10.20~4b763892.tar.zst

重命名、转换格式：
mv libubox-2024.03.29~eb9bcb64.tar.xz libubox-2024-03-29-eb9bcb64.tar.xz
zstd -d -c firmware-utils-2024.10.20~4b763892.tar.zst | xz -z > firmware-utils-2024.10.20-4b763892.tar.xz
zstd -d -c libiwinfo-2025.02.06~9cec6b4d.tar.zst | xz -z > libiwinfo-2025-02-06-9cec6b4d.tar.xz
zstd -d -c rpcd-2024.12.02~cc9a471c.tar.zst | xz -z > rpcd-2024-12-02-cc9a471c.tar.xz
zstd -d -c ubus-2024.10.20~252a9b0c.tar.zst | xz -z > ubus-2024-10-20-252a9b0c.tar.xz

jackzzg, 2025-05-08:
曲线救国试试看能不能成功。

aiamadeus, 2025-05-08:
[image]
你网络问题，不是只有一个下载地址

coolveyron, 2025-05-08:
> > [image] 你网络问题，不是只有一个下载地址
> 
> 用香港、新加坡、硅谷的服务器都出现了同样的问题。

腾讯云服务器吗？腾讯云就别想了，都有这个问题，给他们发了N次工单，答复就是 就这样了，没办法。

jackzzg, 2025-05-09:
> > > [image] 你网络问题，不是只有一个下载地址
> > 
> > 
> > 用香港、新加坡、硅谷的服务器都出现了同样的问题。
> 
> 腾讯云服务器吗？腾讯云就别想了，都有这个问题，给他们发了N次工单，答复就是 就这样了，没办法。

真的是腾讯云的锅，我用阿里云的机器一点问题都没有。

closed 2025-05-09 by coolsnowwolf