coolsnowwolf/lede #13471: ERROR: package/qca/qca-nss-clients failed to build

aNzTikTok, 2025-05-09:
### 详细叙述

ERROR: package/qca/qca-nss-clients failed to build

Full log: https://github.com/aNzTikTok/bin-nss/actions/runs/14937083370/job/41966945646

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

IPQ807X ARCADYAN AW1000

### 详细日志

make[3] -C package/lean/ddns-scripts_dnspod compile
 make[3] -C package/libs/udebug compile
 make[3] -C package/network/config/firewall4 compile
 make[3] -C package/network/services/odhcpd compile
 make[3] -C package/qca/qca-nss-clients compile
    ERROR: package/qca/qca-nss-clients failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/runner/work/bin-nss/bin-nss/openwrt/include/toplevel.mk:231: world] Error 1
Error: Process completed with exit code 2.

coolsnowwolf, 2025-05-10:
只有你自己看得到

coolsnowwolf, 2025-05-12:
推测是不支持 FW4 的 NSS

closed 2025-06-23 by coolsnowwolf