coolsnowwolf/lede #13472: 咨询，选的armv8，编译出来的armv7

binge8, 2025-05-10:
### 详细叙述

现在armv8到底怎么选呢？选的明明是armv8，编译出来的固件名和路径看上去好像armv7呢

[image]
路径和固件名是这样的 openwrt/bin/targets/armsr/armv7/openwrt-armsr-armv7-generic-rootfs.tar.gz

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

armv8

### 详细日志

无

binge8, 2025-05-10:
这个害我盘查了好久才发现，一直扣代码在armsr/armv8路径里找，一直报错，最后发现在armsr/armv7路径

coolsnowwolf, 2025-05-10:
[image]

binge8, 2025-05-11:
> [image]

[image]
我云编译，路径和固件名还是这样，多次确认了，按照我一楼上面那个截图选择的型号，等我本地编译看看是不是吧

binge8, 2025-05-11:
现在还有个报错，从日志上看路径也是armv7
```bash
2025-05-11T18:42:59.0344916Z make[2]: Leaving directory '/home/runner/work/op/op/openwrt'
2025-05-11T18:42:59.1008638Z make[2]: Entering directory '/home/runner/work/op/op/openwrt'
2025-05-11T18:43:01.0289707Z make[3]: Entering directory '/home/runner/work/op/op/openwrt/target/linux'
2025-05-11T18:43:01.0421490Z make[4]: Entering directory '/home/runner/work/op/op/openwrt/target/linux/armsr'
2025-05-11T18:43:01.1250980Z rm -f /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/localversion
2025-05-11T18:43:01.1274476Z /home/runner/work/op/op/openwrt/scripts/kconfig.pl  + + /home/runner/work/op/op/openwrt/target/linux/generic/config-6.6 /home/runner/work/op/op/openwrt/target/linux/armsr/config-6.6 /home/runner/work/op/op/openwrt/target/linux/armsr/armv7/config-6.6 > /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.target
2025-05-11T18:43:01.1680331Z awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/runner/work/op/op/openwrt/.config >> /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.target
2025-05-11T18:43:01.1743505Z echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.target
2025-05-11T18:43:01.1762024Z echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.target
2025-05-11T18:43:01.1782119Z echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.target
2025-05-11T18:43:01.1800339Z /home/runner/work/op/op/openwrt/scripts/package-metadata.pl kconfig /home/runner/work/op/op/openwrt/tmp/.packageinfo /home/runner/work/op/op/openwrt/.config 6.6 > /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.override
2025-05-11T18:43:01.2795235Z /home/runner/work/op/op/openwrt/scripts/kconfig.pl 'm+' '+' /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.target /dev/null /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.override > /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.set
2025-05-11T18:43:01.3256818Z mv /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.set /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.old
2025-05-11T18:43:01.3286205Z grep -v INITRAMFS /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.old > /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.set
2025-05-11T18:43:01.3320316Z echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.set
2025-05-11T18:43:01.3336362Z echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.set
2025-05-11T18:43:01.3353998Z echo "# CONFIG_INITRAMFS_PRESERVE_MTIME is not set" >> /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.set
2025-05-11T18:43:01.3370139Z rm -rf /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/modules
2025-05-11T18:43:01.3395119Z cmp -s /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.set /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.prev || { cp /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.set /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config; cp /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.set /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/.config.prev; }
2025-05-11T18:43:01.3422221Z export MAKEFLAGS= ; [ -d /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90/user_headers ] || make -C /home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-armsr_armv7/linux-6.6.90 KCFLAGS="-fmacro-prefix-map=/home/runner/work/op/op/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi=target-arm_cortex-a15+neon-vfpv4_musl_eabi -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/runner/work/op/op/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_C [... cut ...]
2025-05-11T18:43:42.0194643Z     ERROR: target/linux failed to build.
2025-05-11T18:43:42.0198639Z make[2]: *** [target/Makefile:30: target/linux/install] Error 1
2025-05-11T18:43:42.0203777Z make[2]: Leaving directory '/home/runner/work/op/op/openwrt'
2025-05-11T18:43:42.0238187Z make[1]: *** [target/Makefile:24: /home/runner/work/op/op/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.target_install] Error 2
2025-05-11T18:43:42.0242626Z make[1]: Leaving directory '/home/runner/work/op/op/openwrt'
```

aiamadeus, 2025-05-12:
你需要 rm .config 重新选。。。

binge8, 2025-05-12:
好像没别人遇到我这个问题，那应该是我自己的问题，我自己再盘查看看吧

binge8, 2025-05-12:
确实是config有问题，感谢 @aiamadeus