coolsnowwolf/lede #13491: 小白请教一个关于LuCi-app-软件，排序的问题

5i177, 2025-05-20:
### 详细叙述

用LEDE源码编译添加常用软件后发现软件全堆在服务里面，想请教下前辈如何在编译前修改，让LuCi-app-软件 按字母开头排序，比如 添加了luci-app-passwall luci-app-passwall2 想让它们在服务里面按字母排序，该如何实现。谢谢

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

X86

### 详细日志

用LEDE源码编译添加常用软件后发现软件全堆在服务里面，想请教下前辈如何在编译前修改，让LuCi-app-软件 按字母开头排序，比如 添加了luci-app-passwall luci-app-passwall2 想让它们在服务里面按字母排序，该如何实现。谢谢

5i177, 2025-05-20:
> lua 的 controller 里面可以指定顺序

lua 的 controller 里面可以指定顺序，具体是哪个文件？

liangyuan64, 2025-05-21:
> > lua 的控制器里面可以指定顺序
> 
> 能否全局指定顺序？修改哪个文件？谢谢

/usr/lib/lua/luci/controller/ 修改里面的每一个文件

coolsnowwolf, 2026-09-05:
LuCI 菜单顺序不是按软件包名字自动排序，而是由各个包的菜单定义（`root/usr/share/luci/menu.d/*.json` 的 `order`，旧式页面则由 controller 的 entry order）决定。当前仓库没有统一全局按字母排序的配置项；需要逐个调整菜单定义的 `order`，且不同插件可能属于不同父菜单。原 issue 已有 maintainer 说明该做法，故关闭。

closed 2026-09-05 by coolsnowwolf