coolsnowwolf/lede #13506: Turbo ACC 网络加速设置失败，显示禁用

audyfan, 2025-05-27:
### 详细叙述

mediatek，最新源码编译，内核6.6，编译并使用homeproxy，Turbo ACC 网络加速设置失败。改为编辑并使用openclash时无此问题

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

h3c nx30 pro

### 详细日志

[image]

shjy223534, 2025-05-27:
homeproxy会使用fw4,和Turbo ACC 冲突，编译去掉homeproxy就好了

audyfan, 2025-05-27:
> homeproxy会使用fw4,和Turbo ACC冲突，顺便去掉homeproxy就可以了

感谢回复。原来用的openclash，没有这个问题。现在oc升级越来越占用内存，所以才换了homeproxy的

audyfan, 2025-07-26:
> > homeproxy会使用fw4,和Turbo ACC冲突，顺便去掉homeproxy就可以了
> 
> > homeproxy会使用fw4,和Turbo ACC冲突，顺便去掉homeproxy就可以了
> 
> 我咋感觉好像缺模块呢...

搜了下，是大雕这个turbo不支持fw4

xiandu999-yu, 2025-07-26:
> > > homeproxy会使用fw4,和Turbo ACC冲突，顺便去掉homeproxy就可以了
> > 
> > 
> > > homeproxy会使用fw4,和Turbo ACC冲突，顺便去掉homeproxy就可以了
> > 
> > 
> > 我咋感觉好像缺模块呢...
> 
> 搜了下，是大雕这个turbo不支持fw4

so？

coolsnowwolf, 2026-09-05:
已确认根因是 HomeProxy 使用 fw4，而旧 TurboACC 逻辑不兼容；当前 luci-app-turboacc 已支持 fw4/fullcone 状态检测（RPC 同时检查 nft ruleset 与 `nft_fullcone.ko`），但仍需确保选择对应 nft/fullcone 内核模块。原 issue 无完整配置/日志，现关闭。

closed 2026-09-05 by coolsnowwolf