coolsnowwolf/lede #13524: R6S最新仓库源码编译错误

xiandu999-yu, 2025-05-30:
### 详细叙述

从AI处得到的答案
[image]

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

FriendlyElec NanoPi R6S

### 详细日志

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/002-clk-rockchip-mark-hclk-vi-as-critical-on-rk3568.patch using plaintext:
patching file drivers/clk/rockchip/clk-rk3568.c

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/005-arm64-dts-rockchip-add-EEPROM-node-for-NanoPi-R4S.patch using plaintext:
patching file arch/arm64/boot/dts/rockchip/rk3399-nanopi-r4s.dts

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/010-v6.13-arm64-dts-rockchip-Add-rk3576-SoC-base-DT.patch using plaintext:
patching file arch/arm64/boot/dts/rockchip/rk3576-pinctrl.dtsi
patching file arch/arm64/boot/dts/rockchip/rk3576.dtsi

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/011-v6.13-phy-rockchip-inno-usb2-convert-clock-management-to-bulk.patch using plaintext:
patching file drivers/phy/rockchip/phy-rockchip-inno-usb2.c

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/012-v6.13-phy-rockchip-inno-usb2-Add-usb2-phys-support-for-rk3576.patch using plaintext:
patching file drivers/phy/rockchip/phy-rockchip-inno-usb2.c

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/013-v6.13-phy-rockchip-usbdp-add-rk3576-device-match-data.patch using plaintext:
patching file drivers/phy/rockchip/phy-rockchip-usbdp.c
Hunk https://github.com/coolsnowwolf/lede/issues/1 succeeded at 1556 (offset 19 lines).
Hunk https://github.com/coolsnowwolf/lede/issues/2 succeeded at 1640 (offset 19 lines).

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/014-v6.13-gpio-rockchip-explan-the-format-of-the-GPIO-version-ID.patch using plaintext:
patching file drivers/gpio/gpio-rockchip.c

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/015-v6.13-gpio-rockchip-change-the-GPIO-version-judgment-logic.patch using plaintext:
patching file drivers/gpio/gpio-rockchip.c

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/016-v6.13-gpio-rockchip-support-new-version-GPIO.patch using plaintext:
patching file drivers/gpio/gpio-rockchip.c

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/020-v6.13-irqchip-gic-v3-its-Share-ITS-tables-with-a-non-trust.patch using plaintext:
patching file drivers/irqchip/irq-gic-v3-its.c

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/021-v6.13-irqchip-gic-v3-its-Fix-over-allocation-in-itt_alloc.patch using plaintext:
patching file drivers/irqchip/irq-gic-v3-its.c

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/022-v6.15-irqchip-gic-v3-Add-Rockchip-3568002-erratum-workaround.patch using plaintext:
patching file Documentation/arch/arm64/silicon-errata.rst
patching file arch/arm64/Kconfig
patching file drivers/irqchip/irq-gic-v3-its.c

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/023-01-v6.15-arm64-dts-rockchip-rk356x-Add-MSI-controller-node.patch using plaintext:
patching file arch/arm64/boot/dts/rockchip/rk356x.dtsi

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/023-02-v6.15-arm64-dts-rockchip-rk356x-Move-PCIe-MSI-to-use-GIC.patch using plaintext:
patching file arch/arm64/boot/dts/rockchip/rk356x.dtsi

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/100-rockchip-use-system-LED-for-OpenWrt.patch using plaintext:
patching file arch/arm64/boot/dts/rockchip/rk3328-nanopi-r2s.dts
patching file arch/arm64/boot/dts/rockchip/rk3328-roc-cc.dts

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/101-net-realtek-r8169-add-LED-configuration-from-OF.patch using plaintext:
patching file drivers/net/ethernet/realtek/r8169_main.c
Hunk https://github.com/coolsnowwolf/lede/issues/2 succeeded at 242 (offset 66 lines).
Hunk https://github.com/coolsnowwolf/lede/issues/3 succeeded at 5380 with fuzz 1 (offset 7 lines).
Hunk https://github.com/coolsnowwolf/lede/pull/4 succeeded at 5567 (offset 10 lines).

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/103-arm64-rockchip-add-OF-node-for-USB-eth-on-NanoPi-R2S.patch using plaintext:
patching file arch/arm64/boot/dts/rockchip/rk3328-nanopi-r2s.dts

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/105-nanopi-r4s-sd-signalling.patch using plaintext:
patching file arch/arm64/boot/dts/rockchip/rk3328-roc-cc.dts
patching file arch/arm64/boot/dts/rockchip/rk3399-nanopi-r4s.dts

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/106-r4s-openwrt-leds.patch using plaintext:
patching file arch/arm64/boot/dts/rockchip/rk3399-nanopi-r4s.dts

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/107-mmc-core-set-initial-signal-voltage-on-power-off.patch using plaintext:
patching file drivers/mmc/core/core.c

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/110-arm64-rk3568-update-gicv3-its-and-pci-msi-map.patch using plaintext:
patching file arch/arm64/boot/dts/rockchip/rk3568.dtsi

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/112-arm64-dts-rockchip-rk3568-Add-xpcs-support.patch using plaintext:
patching file arch/arm64/boot/dts/rockchip/rk3568.dtsi
patching file arch/arm64/boot/dts/rockchip/rk356x.dtsi

Applying /home/xianyu/lede/target/linux/rockchip/patches-6.12/113-ethernet-stmicro-stmmac-Add-SGMII-QSGMII-support.patch using plaintext:
patching file drivers/net/ethernet/stmicro/stmmac/dwmac-rk.c
Hunk https://github.com/coolsnowwolf/lede/issues/3 succeeded at 42 (offset 1 line).
Hunk https://github.com/coolsnowwolf/lede/pull/4 succeeded at 52 (offset 1 line).
Hunk https://github.com/coolsnowwolf/lede/issues/5 succeeded at 84 (offset 1 line).
Hunk https://github.com/coolsnowwolf/lede/pull/6 succeeded at 97 (offset 1 line).
Hunk https://github.com/coolsnowwolf/lede/issues/7 succeeded at 1147 (offset 1 line).
Hunk https://github.com/coolsnowwolf/lede/issues/8 succeeded at 1163 (offset 1 line).
Hunk https://github.com/coolsnowwolf/lede/issues/9 succeeded at 1275 (offset 1 line).
Hunk https://github.com/coolsnowwolf/lede/issues/10 succeeded at 1906 (offset 3 lines).
Hunk https://github.com/coolsnowwolf/lede/issues/11 FAILED at 2002.
Hunk https://github.com/coolsnowwolf/lede/pull/12 succeeded at 2096 (offset 17 lines).
Hunk https://github.com/coolsnowwolf/lede/issues/13 succeeded at 2129 (offset 17 lines).
Hunk https://github.com/coolsnowwolf/lede/issues/14 succeeded at 2150 (offset 17 lines).
1 out of 14 hunks FAILED -- saving rejects to file drivers/net/ethernet/stmicro/stmmac/dwmac-rk.c.rej
Patch failed! Please fix /home/xianyu/lede/target/linux/rockchip/patches-6.12/113-ethernet-stmicro-stmmac-Add-SGMII-QSGMII-support.patch!
make[4]: *** [Makefile:34: /home/xianyu/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.12.31/.prepared_c601c7f7cdac09d136342ec9518fcc35] Error 1
make[4]: Leaving directory '/home/xianyu/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/xianyu/lede/target/linux'
time: target/linux/compile#9.01#11.03#15.13
ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/xianyu/lede'
make[1]: *** [target/Makefile:23: /home/xianyu/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/xianyu/lede'
make: *** [/home/xianyu/lede/include/toplevel.mk:231: world] Error 2

coolsnowwolf, 2025-05-30:
问 AI 好像没啥用？ 修复了

closed 2025-05-30 by coolsnowwolf

xiandu999-yu, 2025-05-31:
> 问 AI 好像没啥用？ 修复了

> 问 AI 好像没啥用？ 修复了

问问也没啥坏处