coolsnowwolf/lede #13526: x86最新源码无法编译

Rinfair, 2025-05-31:
### 详细叙述

初次编译失败后剔除了AppArmor相关包，后仍然编译失败（此次编译失败无法处理，因为出错的包为必需）

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

R5-8645HS+Intel I226-V+Intel AX200工程板

### 详细日志

```  
  CC [M]  /home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/coex.o
  CC [M]  /home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.o
/home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c: In function 'iwl_mvm_tzone_get_temp':
/home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:618:55: error: invalid use of undefined type 'struct thermal_zone_device'
  618 |         struct iwl_mvm *mvm = (struct iwl_mvm *)device->devdata;
      |                                                       ^~
/home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c: In function 'iwl_mvm_tzone_get_trip_temp':
/home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:644:55: error: invalid use of undefined type 'struct thermal_zone_device'
  644 |         struct iwl_mvm *mvm = (struct iwl_mvm *)device->devdata;
      |                                                       ^~
/home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c: In function 'iwl_mvm_tzone_set_trip_temp':
/home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:668:55: error: invalid use of undefined type 'struct thermal_zone_device'
  668 |         struct iwl_mvm *mvm = (struct iwl_mvm *)device->devdata;
      |                                                       ^~
/home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c: At top level:
/home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:723:10: error: 'struct thermal_zone_device_ops' has no member named 'get_trip_temp'; did you mean 'set_trip_temp'?
  723 |         .get_trip_temp = iwl_mvm_tzone_get_trip_temp,
      |          ^~~~~~~~~~~~~
      |          set_trip_temp
/home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:723:26: error: initialization of 'int (*)(struct thermal_zone_device *, int,  int)' from incompatible pointer type 'int (*)(struct thermal_zone_device *, int,  int *)' [-Werror=incompatible-pointer-types]
  723 |         .get_trip_temp = iwl_mvm_tzone_get_trip_temp,
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:723:26: note: (near initialization for 'tzone_ops.set_trips')
/home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:724:10: error: 'struct thermal_zone_device_ops' has no member named 'get_trip_type'; did you mean 'set_trip_temp'?
  724 |         .get_trip_type = iwl_mvm_tzone_get_trip_type,
      |          ^~~~~~~~~~~~~
      |          set_trip_temp
/home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:724:26: error: initialization of 'int (*)(struct thermal_zone_device *, enum thermal_device_mode)' from incompatible pointer type 'int (*)(struct thermal_zone_device *, int,  enum thermal_trip_type *)' [-Werror=incompatible-pointer-types]
  724 |         .get_trip_type = iwl_mvm_tzone_get_trip_type,
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:724:26: note: (near initialization for 'tzone_ops.change_mode')
/home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:725:26: error: initialization of 'int (*)(struct thermal_zone_device *, const struct thermal_trip *, int)' from incompatible pointer type 'int (*)(struct thermal_zone_device *, int,  int)' [-Werror=incompatible-pointer-types]
  725 |         .set_trip_temp = iwl_mvm_tzone_set_trip_temp,
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:725:26: note: (near initialization for 'tzone_ops.set_trip_temp')
/home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c: In function 'iwl_mvm_thermal_zone_register':
/home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:746:32: error: implicit declaration of function 'thermal_zone_device_register'; did you mean 'thermal_zone_device_unregister'? [-Werror=implicit-function-declaration]
  746 |         mvm->tz_device.tzone = thermal_zone_device_register(name,
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                thermal_zone_device_unregister
/home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:746:30: warning: assignment to 'struct thermal_zone_device *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
  746 |         mvm->tz_device.tzone = thermal_zone_device_register(name,
      |                              ^
cc1: some warnings being treated as errors
make[12]: *** [scripts/Makefile.build:229: /home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.o] Error 1
make[11]: *** [scripts/Makefile.build:478: /home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi/mvm] Error 2
make[10]: *** [scripts/Makefile.build:478: /home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel/iwlwifi] Error 2
make[9]: *** [scripts/Makefile.build:478: /home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless/intel] Error 2
make[8]: *** [scripts/Makefile.build:478: /home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/drivers/net/wireless] Error 2
make[7]: *** [Makefile:1945: /home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:93: modules] Error 2
make[4]: *** [Makefile:120: modules] Error 2
make[4]: Leaving directory '/home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1'
make[3]: *** [Makefile:396: /home/rinfair/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.110-1/.built] Error 2
make[3]: Leaving directory '/home/rinfair/lede/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#21.41#3.84#25.07
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/rinfair/lede'
make[1]: *** [package/Makefile:110: /home/rinfair/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/rinfair/lede'
make: *** [/home/rinfair/lede/include/toplevel.mk:231：world] 错误 2
```

mmeiwenlong, 2025-05-31:
[image]

Rinfair, 2025-05-31:
> [image] [doge]

`make menuconfig`没有kmod-thermal

closed 2025-06-04 by coolsnowwolf