coolsnowwolf/lede #13527: x86 J4125 最新源码编译固件仍有开机时因i915超时卡死问题

MilesPoupart, 2025-05-31:
### 详细叙述

已使用添加了i915的最新版本源码

[image]

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

x86 J4125

### 详细日志

内核日志
```
[    3.560046] i915 0000:00:02.0: [drm] Unknown revision 0x06
[    3.565557] i915 0000:00:02.0: [drm] Found GEMINILAKE (device ID 3185) display version 10.00 stepping **
[    3.575048] i915 0000:00:02.0: [drm] Unknown revid 0x06
[    3.581176] i915 0000:00:02.0: [drm] couldn't get memory information
[    3.588586] i915 0000:00:02.0: Direct firmware load for i915/glk_dmc_ver1_04.bin failed with error -2
[    3.597832] i915 0000:00:02.0: Falling back to sysfs fallback for: i915/glk_dmc_ver1_04.bin
[   66.085277] i915 0000:00:02.0: [drm] *ERROR* GT0: GuC firmware i915/glk_guc_70.1.1.bin: fetch failed -ETIMEDOUT
[   66.085281] i915 0000:00:02.0: [drm] Failed to load DMC firmware i915/glk_dmc_ver1_04.bin (-ETIMEDOUT). Disabling runtime power management.
[   66.095379] i915 0000:00:02.0: [drm] GT0: GuC firmware(s) can be downloaded from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915
[   66.107902] i915 0000:00:02.0: [drm] DMC firmware homepage: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
[   66.123713] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/glk_guc_70.1.1.bin version 0.0.0
[   66.143498] i915 0000:00:02.0: [drm] GT0: GuC is uninitialized
[   66.150865] [drm] Initialized i915 1.6.0 for 0000:00:02.0 on minor 0
[   66.158386] ACPI: video: Video Device [GFX0] (multi-head: yes  rom: no  post: no)
[   66.166178] input: Video Bus as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/LNXVIDEO:00/input/input1
[   66.195269] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[   66.216623] loop: module loaded
[   66.219808] Guest personality initialized and is inactive
[   66.225249] VMCI host device registered (name=vmci, major=10, minor=127)
[   66.231969] Initialized host personality
[   66.235241] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[   66.235966] lpc_ich 0000:00:1f.0: No MFD cells added
[   66.247172] VMware PVSCSI driver - version 1.0.7.0-k
[   66.252166] hv_vmbus: registering driver hv_storvsc
[   66.255264] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
```

MilesPoupart, 2025-06-01:
[image]
看了下/lib/firmware/i915/里面文件是有的

aiamadeus, 2025-06-02:
要把 x86 kconfig 的 i915 去掉
kmod 模块才能加载 firmware

closed 2025-06-04 by MilesPoupart