coolsnowwolf/lede #13536: AP4220 编译报错， ERROR: package/kernel/mac80211 failed to build (build variant: regular).

kavass168, 2025-06-04:
### 详细叙述

ERROR: package/kernel/mac80211 failed to build (build variant: regular).

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

AP4220

### 详细日志

make[3] -C feeds/packages/libs/libffi host-compile
 make[3] -C package/utils/bzip2 host-compile
 make[3] -C feeds/packages/lang/python/python3 host-compile
 make[3] -C feeds/packages/libs/rpcsvc-proto host-compile
 make[3] -C feeds/packages/net/samba4 host-compile
 make[3] -C package/libs/popt compile
 make[3] -C feeds/packages/net/samba4 compile
 make[3] -C feeds/packages/net/wget compile
 make[3] -C feeds/packages/net/wsdd2 compile
 make[3] -C package/firmware/wireless-regdb compile
 make[3] -C package/network/services/hostapd compile
 make[3] -C package/network/utils/iw compile
 make[3] -C package/kernel/mac80211 compile
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/workdir/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/workdir/openwrt/scripts/config'
make[1]: Entering directory '/workdir/openwrt'
make[2]: Entering directory '/workdir/openwrt'
make[3]: Entering directory '/workdir/openwrt/target/linux'
make[4]: Entering directory '/workdir/openwrt/target/linux/ipq40xx

MilesPoupart, 2025-06-04:
没看到详细日志 我这里6.12 x86编译mac80211也有问题 不知道是不是同一个问题
https://github.com/coolsnowwolf/lede/issues/13513#issuecomment-2936092401

coolsnowwolf, 2025-06-05:
已修复