coolsnowwolf/lede #13543: 希望添加CMCC RAX3000Me支持

xiaoliangs, 2025-06-06:
### 详细叙述

希望LEDE可以支持CMCC RAX3000Me这款设备，感谢。

Immortalwrt已经支持这台设备了：
https://firmware-selector.immortalwrt.org/?version=24.10.1&target=mediatek%2Ffilogic&id=cmcc_rax3000me

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

CMCC RAX3000Me

### 详细日志

无。

Kuron88, 2025-06-08:
这个本来就支持啊

xiaoliangs, 2025-06-08:
> 这个本来就支持啊

你看我发的编译截图，没有这个型号

coolsnowwolf, 2025-06-13:
固件通刷

coolsnowwolf, 2025-06-23:
添加支持了

chr09670, 2025-06-26:
现在可以支持了吗？刚入了这个设备

coolsnowwolf, 2026-09-05:
已核对当前 LEDE master：现有 `cmcc_rax3000m-emmc` / `cmcc_rax3000m-nand` 仅定义 RAX3000M，仓库中没有 `cmcc_rax3000me` 的独立 DTS、无 USB overlay 或 DDR3-1866 启动介质定义；因此不能仅凭“RAX3000M 固件通刷”确认 RAX3000Me 可启动。OpenWrt/ImmortalWrt 的完整支持包含额外 U-Boot、DTS overlay 和 DDR3 启动介质，当前无法在没有设备实测的情况下安全压缩移植。该 issue 暂不关闭，待按完整上游支持移植并在硬件上验证。

coolsnowwolf, 2026-09-05:
已补充 CMCC RAX3000Me 支持并推送：

- `1397f1c98ff`：加入 RAX3000Me 的 Linux DTS、NAND/eMMC 镜像定义、DDR3/DDR4/DDR3-1866 ATF 与 U-Boot 变体、升级及 Wi-Fi MAC 处理。
- `f81e0a69a13`：补齐当前分支缺失的 RAX3000M U-Boot 基础 DTS/defconfig/env patch，确保 Me DTS 的 include 链完整。

验证：U-Boot prepare 成功，aarch64 目标编译已通过 RAX3000Me DTB 的 DTC 阶段；当前仓库 x86_64 `.config` 的顶层编译不能代表 ARM 目标，直接链接阶段受现有 U-Boot board_init 配置限制。

closed 2026-09-05 by coolsnowwolf