coolsnowwolf/lede #13553: 编译 6.1内核的 arm64 qemu 报错没有libphy.ko phylink.ko 依赖

Border-cu, 2025-06-15:
### 详细叙述

make menuconfig 界面已经勾选了libphy.ko  phylink.ko 这两个内核模块，无论是编译6.1.137内核的 纯净SDK和编译带有openclash的固件 都有以上报错，5.15的没问题

可以参考一下这两个issues
https://github.com/coolsnowwolf/lede/issues/8669
https://github.com/openwrt/openwrt/issues/12758

正在尝试修改netdevice.mk解决

define KernelPackage/pcs-xpcs
  SUBMENU:=$(NETWORK_DEVICES_MENU)
  TITLE:=Synopsis DesignWare PCS driver
  DEPENDS:=@(TARGET_x86_64||TARGET_armsr||TARGET_armvirt_64) +kmod-libphy +kmod-phylink +LINUX_6_12:kmod-mdio-devres
  KCONFIG:=CONFIG_PCS_XPCS
  FILES:=$(LINUX_DIR)/drivers/net/pcs/pcs_xpcs.ko
  AUTOLOAD:=$(call AutoLoad,20,pcs_xpcs)
endef

$(eval $(call KernelPackage,pcs-xpcs))

另外 内核版本高于6.1.137 的 target system中都没有qemu arm 

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

arm64 QEMU

### 详细日志

rstrip.sh: /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/packages/ipkg-aarch64_generic/kmod-tun/lib/modules/6.1.137/tun.ko: relocatable
Packaged contents of /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/packages/ipkg-aarch64_generic/kmod-tun into /home/zxy/lede/bin/targets/armvirt/64/packages/kmod-tun_6.1.137-1_aarch64_generic.ipk
rstrip.sh: /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/packages/ipkg-aarch64_generic/kmod-slhc/lib/modules/6.1.137/slhc.ko: relocatable
Packaged contents of /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/packages/ipkg-aarch64_generic/kmod-slhc into /home/zxy/lede/bin/targets/armvirt/64/packages/kmod-slhc_6.1.137-1_aarch64_generic.ipk
rstrip.sh: /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/packages/ipkg-aarch64_generic/kmod-ppp/lib/modules/6.1.137/ppp_async.ko: relocatable
rstrip.sh: /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/packages/ipkg-aarch64_generic/kmod-ppp/lib/modules/6.1.137/ppp_generic.ko: relocatable
Packaged contents of /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/packages/ipkg-aarch64_generic/kmod-ppp into /home/zxy/lede/bin/targets/armvirt/64/packages/kmod-ppp_6.1.137-1_aarch64_generic.ipk
rstrip.sh: /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/packages/ipkg-aarch64_generic/kmod-pppox/lib/modules/6.1.137/pppox.ko: relocatable
Packaged contents of /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/packages/ipkg-aarch64_generic/kmod-pppox into /home/zxy/lede/bin/targets/armvirt/64/packages/kmod-pppox_6.1.137-1_aarch64_generic.ipk
rstrip.sh: /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/packages/ipkg-aarch64_generic/kmod-pppoe/lib/modules/6.1.137/pppoe.ko: relocatable
Packaged contents of /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/packages/ipkg-aarch64_generic/kmod-pppoe into /home/zxy/lede/bin/targets/armvirt/64/packages/kmod-pppoe_6.1.137-1_aarch64_generic.ipk
rstrip.sh: /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/packages/ipkg-aarch64_generic/kmod-mppe/lib/modules/6.1.137/ppp_mppe.ko: relocatable
Packaged contents of /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/packages/ipkg-aarch64_generic/kmod-mppe into /home/zxy/lede/bin/targets/armvirt/64/packages/kmod-mppe_6.1.137-1_aarch64_generic.ipk
rstrip.sh: /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/packages/ipkg-aarch64_generic/kmod-tcp-bbr/lib/modules/6.1.137/tcp_bbr.ko: relocatable
Packaged contents of /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/packages/ipkg-aarch64_generic/kmod-tcp-bbr into /home/zxy/lede/bin/targets/armvirt/64/packages/kmod-tcp-bbr_6.1.137-1_aarch64_generic.ipk
rstrip.sh: /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/packages/ipkg-aarch64_generic/kmod-rtc-rx8025/lib/modules/6.1.137/rtc-rx8025.ko: relocatable
Packaged contents of /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/packages/ipkg-aarch64_generic/kmod-rtc-rx8025 into /home/zxy/lede/bin/targets/armvirt/64/packages/kmod-rtc-rx8025_6.1.137-1_aarch64_generic.ipk
Packaged contents of /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/packages/ipkg-aarch64_generic/kmod-acpi-mdio into /home/zxy/lede/bin/targets/armvirt/64/packages/kmod-acpi-mdio_6.1.137-1_aarch64_generic.ipk
Package kmod-pcs-xpcs is missing dependencies for the following libraries:
libphy.ko
phylink.ko
make[3]: *** [/home/zxy/lede/target/linux/armvirt/modules.mk:25: /home/zxy/lede/bin/targets/armvirt/64/packages/kmod-pcs-xpcs_6.1.137-1_aarch64_generic.ipk] Error 1
make[3]: Leaving directory '/home/zxy/lede/package/kernel/linux'
time: package/kernel/linux/compile#9.36#12.94#19.38
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/zxy/lede'
make[1]: *** [package/Makefile:110: /home/zxy/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zxy/lede'
make: *** [/home/zxy/lede/include/toplevel.mk:231: world] Error 2

aiamadeus, 2025-06-15:
早就没 armvirt 了，不知道你编译的什么上古版本

Border-cu, 2025-06-15:
https://github.com/coolsnowwolf/lede/commit/4f5a34e801dd41fe9d4125e7caa5056aa6575814
emmmm，这个版本，那我要是想编译用于arm64 的  6.1内核 的虚拟机或者tar.gz 应该选择什么Target System，大佬能指点一下吗

Border-cu, 2025-06-15:
大佬 我又用最新的 https://github.com/coolsnowwolf/lede/commit/ea83216ded8cbf553495e4bf48fd3d99b1f31762 编译的6.1.141内核armv8  ，然后报错

```
rstrip.sh: /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armsr_armv8/packages/ipkg-aarch64_generic/kmod-asn1-decoder/lib/modules/6.1.141/asn1_decoder.ko: relocatable
Packaged contents of /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armsr_armv8/packages/ipkg-aarch64_generic/kmod-asn1-decoder into /home/zxy/lede/bin/targets/armsr/armv8/packages/kmod-asn1-decoder_6.1.141-1_aarch64_generic.ipk
rstrip.sh: /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armsr_armv8/packages/ipkg-aarch64_generic/kmod-libphy/lib/modules/6.1.141/libphy.ko: relocatable
Packaged contents of /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armsr_armv8/packages/ipkg-aarch64_generic/kmod-libphy into /home/zxy/lede/bin/targets/armsr/armv8/packages/kmod-libphy_6.1.141-1_aarch64_generic.ipk
rstrip.sh: /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armsr_armv8/packages/ipkg-aarch64_generic/kmod-phylink/lib/modules/6.1.141/phylink.ko: relocatable
Packaged contents of /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armsr_armv8/packages/ipkg-aarch64_generic/kmod-phylink into /home/zxy/lede/bin/targets/armsr/armv8/packages/kmod-phylink_6.1.141-1_aarch64_generic.ipk
rstrip.sh: /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armsr_armv8/packages/ipkg-aarch64_generic/kmod-mii/lib/modules/6.1.141/mii.ko: relocatable
Packaged contents of /home/zxy/lede/build_dir/target-aarch64_generic_musl/linux-armsr_armv8/packages/ipkg-aarch64_generic/kmod-mii into /home/zxy/lede/bin/targets/armsr/armv8/packages/kmod-mii_6.1.141-1_aarch64_generic.ipk
Package kmod-mdio-devres is missing dependencies for the following libraries:
of_mdio.ko
make[3]: *** [modules/netdevices.mk:157: /home/zxy/lede/bin/targets/armsr/armv8/packages/kmod-mdio-devres_6.1.141-1_aarch64_generic.ipk] Error 1
make[3]: Leaving directory '/home/zxy/lede/package/kernel/linux'
time: package/kernel/linux/compile#7.44#7.08#12.97
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/zxy/lede'
make[1]: *** [package/Makefile:110: /home/zxy/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zxy/lede'
make: *** [/home/zxy/lede/include/toplevel.mk:231: world] Error 2
```

找到文件
lede/package/kernel/linux/modules/netdevices.mk
的相关部分

```
define KernelPackage/mdio-devres
  SUBMENU:=$(NETWORK_DEVICES_MENU)
  TITLE:=Supports MDIO device registration
  DEPENDS:=@!LINUX_5_4 +kmod-libphy +(TARGET_armvirt||TARGET_bcm27xx_bcm2708||TARGET_loongarch64||TARGET_malta||TARGET_tegra):kmod-of-mdio
  KCONFIG:=CONFIG_MDIO_DEVRES
  HIDDEN:=1
  FILES:=$(LINUX_DIR)/drivers/net/phy/mdio_devres.ko
  AUTOLOAD:=$(call AutoProbe,mdio-devres)
endef

define KernelPackage/mdio-devres/description
 Supports MDIO device registration
endef

$(eval $(call KernelPackage,mdio-devres))
```
依然是TARGET_armvirt 不是armsr 或者 armv8 好几个依赖都是这种情况
这是不是不正常

closed 2025-06-30 by coolsnowwolf