coolsnowwolf/lede #13554: 编译固件默认未选中ipv6但是编译出的固件有ipv6

Lai-xi, 2025-06-15:
### 详细叙述

编译固件默认未选中ipv6helper、dockers等包含ipv6的配置，但是编译出的固件有ipv6。2024年年底反馈过，最近编译发现还存在该问题。

[image]

[image]

[image]

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

MT7981-RAX3000M

### 详细日志

编译无报错

markvlenvision, 2025-06-22:
一样

coolsnowwolf, 2025-06-23:
目前 ipv6 和组件的关联已经很紧密了，随随便便就关联上了

Lai-xi, 2025-06-23:
> 目前 ipv6 和组件的关联已经很紧密了，随随便便就关联上了

感觉是过于紧密了😂，我导入以前老的配置文件，在老的配置的基础上重新选择插件编译出来是没有ipv6的，全新编译相同的插件就带ipv6。

[image]

coolsnowwolf, 2026-09-05:
已核对当前构建系统：`include/target.mk` 的 `DEFAULT_PACKAGES.router` 无条件包含 `odhcp6c`、`odhcpd-ipv6only`、`ip6tables`、`libip6tc`、`kmod-ipt-nat6`、`luci-proto-ipv6` 等 IPv6 组件；因此即使配置界面未单独勾选 ipv6helper，路由器目标仍会带入 IPv6 用户空间/防火墙组件。这是当前默认包策略，不是编译失败或包遗漏。若希望完全去除 IPv6，需要在定制配置中移除这些默认包并确认依赖闭包。原 issue 没有可对应的源码错误，先关闭。

closed 2026-09-05 by coolsnowwolf