coolsnowwolf/lede #13572: 添加ax210固件支持后mac80211报错

Link-Zhang, 2025-06-22:
### 详细叙述

我一直卡在为香橙派5plus添加ax210以使用wifi的路上，其中我用附件的0.0.9编译不带ax210的wifi相关选项时，一切正常。但是使用0.0.10版本去除一些DEFAULT 的选项pi5plus.config后，一直会出各种各样的问题， 比方说磁盘空间不足，或者mac80211编译失败，或者hostapd 编译失败。麻烦大佬帮忙看看，我的自定义固件，就差个wifi功能了。谢谢

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

xunlong orange pi5plus

### 详细日志

make[8]: 'conf' is up to date.
boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
#
# configuration written to .config
#
Building backport-include/backport/autoconf.h ... done.
  CC [M]  /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/drivers/net/wireless/intel/iwlwifi/mvm/tt.o
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:754:26: error: initialization of 'int (*)(struct thermal_zone_device *, const struct thermal_trip *, int)' from incompatible pointer type 'int (*)(struct thermal_zone_device *, int,  int)' [-Werror=incompatible-pointer-types]
  754 |         .set_trip_temp = iwl_mvm_tzone_set_trip_temp,
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:754:26: note: (near initialization for 'tzone_ops.set_trip_temp')
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/drivers/net/wireless/intel/iwlwifi/mvm/tt.c: In function 'iwl_mvm_thermal_zone_register':
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:758:32: warning: passing argument 4 of 'thermal_zone_device_register_with_trips' makes pointer from integer without a cast [-Wint-conversion]
  758 | #define IWL_WRITABLE_TRIPS_MSK (BIT(IWL_MAX_DTS_TRIPS) - 1)
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                |
      |                                long unsigned int
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:782:57: note: in expansion of macro 'IWL_WRITABLE_TRIPS_MSK'
  782 |                                                         IWL_WRITABLE_TRIPS_MSK,
      |                                                         ^~~~~~~~~~~~~~~~~~~~~~
In file included from /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/backport-include/linux/thermal.h:3,
                 from /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/drivers/net/wireless/intel/iwlwifi/mvm/mvm.h:16,
                 from /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:9:
./include/linux/thermal.h:225:62: note: expected 'void *' but argument is of type 'long unsigned int'
  225 |                                         int num_trips, void *devdata,
        |                                                        ~~~~~~^~~~~~~
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:783:57: error: passing argument 5 of 'thermal_zone_device_register_with_trips' from incompatible pointer type [-Werror=incompatible-pointer-types]
  783 |                                                         mvm, &tzone_ops,
      |                                                         ^~~
      |                                                         |
      |                                                         struct iwl_mvm *
./include/linux/thermal.h:226:79: note: expected 'const struct thermal_zone_device_ops *' but argument is of type 'struct iwl_mvm *'
  226 |                                         const struct thermal_zone_device_ops *ops,
      |                                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:783:62: error: passing argument 6 of 'thermal_zone_device_register_with_trips' from incompatible pointer type [-Werror=incompatible-pointer-types]
  783 |                                                         mvm, &tzone_ops,
      |                                                              ^~~~~~~~~~
      |                                                              |
      |                                                              struct thermal_zone_device_ops *
./include/linux/thermal.h:227:75: note: expected 'const struct thermal_zone_params *' but argument is of type 'struct thermal_zone_device_ops *'
  227 |                                         const struct thermal_zone_params *tzp,
      |                                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
In file included from /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/backport-include/linux/stddef.h:3,
                 from ./include/uapi/linux/posix_types.h:5,
                 from ./include/uapi/linux/types.h:14,
                 from ./include/linux/types.h:6,
                 from /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/backport-include/linux/types.h:4,
                 from ./include/linux/sort.h:5,
                 from /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:7:
./include/linux/stddef.h:8:14: warning: passing argument 7 of 'thermal_zone_device_register_with_trips' makes integer from pointer without a cast [-Wint-conversion]
    8 | #define NULL ((void *)0)
      |              ^~~~~~~~~~~
      |              |
      |              void *
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/drivers/net/wireless/intel/iwlwifi/mvm/tt.c:784:57: note: in expansion of macro 'NULL'
  784 |                                                         NULL, 0, 0);
[... 6 lines cut ...]
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./include/linux/thermal.h:222:29: note: declared here
  222 | struct thermal_zone_device *thermal_zone_device_register_with_trips(
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[12]: *** [scripts/Makefile.build:229: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/drivers/net/wireless/intel/iwlwifi/mvm/tt.o] Error 1
make[11]: *** [scripts/Makefile.build:478: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/drivers/net/wireless/intel/iwlwifi/mvm] Error 2
make[10]: *** [scripts/Makefile.build:478: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/drivers/net/wireless/intel/iwlwifi] Error 2
make[9]: *** [scripts/Makefile.build:478: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/drivers/net/wireless/intel] Error 2
make[8]: *** [scripts/Makefile.build:478: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/drivers/net/wireless] Error 2
make[7]: *** [Makefile:1945: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:105: modules] Error 2
make[4]: *** [Makefile:121: modules] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15'
make[3]: *** [Makefile:405: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.6.15/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#1.13#0.49#1.63
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

Link-Zhang, 2025-06-22:
Full 详细日志:

[010.txt](https://github.com/user-attachments/files/20852756/010.txt)

Link-Zhang, 2025-06-23:
> 已修复

大佬，请问下，如何切换内核到5.10啊？

coolsnowwolf, 2025-06-23:
RK3588 在 GPL  5.10 内核是不支持的

TxiRan, 2025-11-12:
> > RK3588 在 GPL 5.10 内核是不支持的
> 
> 不好意思，大佬，再问下，是不是因为RK3588是openwrt 24才支持的，所以RK3588只有在GPL 6.6 + 的内核上才会支持？

请问目前是否有解决该`硬件锁`问题的方案，我也遇到了这个问题