coolsnowwolf/lede #13587: 编译openclash报错请大佬进来指教谢谢~~~~

hztbhtbh, 2025-07-01:
### 详细叙述

各位大佬上午好：
编译环境：虚拟机vmware中安装ubuntu22.04;
编译源码：lean的lede源代码编译X86_64,内核是5.4.234，luci是18.06的。
三方插件：kenzok8：在feeds.conf.default中添加：src-git small https://github.com/kenzok8/small
安装最后一个插件openclash最后报错，单线程编译报错代码如下，昨天捣鼓了一天在chatgpt和谷歌搜索都没有搞定
所以前来讨教。

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

X86_64

### 详细日志

make[3]: Entering directory '/home/tbh/lede/feeds/packages/lang/rust'
CARGO_HOME=/home/tbh/lede/dl/cargo TARGET_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/tbh/lede/build_dir/target-x86_64_musl/rust-1.87.0=rust-1.87.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" python3 /home/tbh/lede/build_dir/target-x86_64_musl/host/rustc-1.87.0-src/x.py --build-dir /home/tbh/lede/build_dir/target-x86_64_musl/host/rustc-1.87.0-src/build --ci false dist build-manifest rustc rust-std cargo llvm-tools rust-src
Building bootstrap
Finished dev profile [unoptimized] target(s) in 0.68s
WARNING: The change-id is missing in the bootstrap.toml. This means that you will not be able to track the major changes made to the bootstrap configurations.
NOTE: to silence this warning, add change-id = 138986 or change-id = "ignore" at the top of bootstrap.toml

thread 'main' panicked at src/bootstrap/src/core/sanity.rs:336:25:
couldn't find libc.a in musl libdir: lib
note: run with RUST_BACKTRACE=1 environment variable to display a backtrace
Build completed unsuccessfully in 0:00:00
make[3]: *** [Makefile:109: /home/tbh/lede/build_dir/target-x86_64_musl/host/rustc-1.87.0-src/.built] Error 1
make[3]: Leaving directory '/home/tbh/lede/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#0.82#0.35#1.16
ERROR: package/feeds/packages/rust [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/rust/host/compile] Error 1
make[2]: Leaving directory '/home/tbh/lede'
make[1]: *** [package/Makefile:110: /home/tbh/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/tbh/lede'
make: *** [/home/tbh/lede/include/toplevel.mk:231: world] Error 2

Internet1235, 2025-07-01:
版本太老了。。

diabolic00, 2025-07-03:
俺也一样,1周了全卡在ss-l的rust上,自己装Ubuntu会卡,但是用github的actions就能正常过.
   ERROR: package/feeds/packages/rust [host] failed to build.

coolsnowwolf, 2025-07-03:
为啥要用 R23 的老版本工具链？ 要老版本 luci 直接修改 feeds.conf.default 就好了

a541619896, 2025-07-04:
我也一样 报错提示都是一毛一样 用的H打的MT798X源码手动添加了55r-rust版就开始有这个问题了https://github.com/hanwckf/immortalwrt-mt798x
WARNING: The `change-id` is missing in the `bootstrap.toml`. This means that you will not be able to track the major changes made to the bootstrap configurations.
NOTE: to silence this warning, add `change-id = 138986` or change-id = "ignore" at the top of `bootstrap.toml`

thread 'main' panicked at src/bootstrap/src/core/sanity.rs:336:25:
couldn't find libc.a in musl libdir: lib
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Build completed unsuccessfully in 0:00:01
make[3]: *** [Makefile:110: /home/builder/immortalwrt-mt798x/build_dir/target-aarch64_cortex-a53_musl/host/rustc-1.87.0-src/.built] Error 1
make[3]: Leaving directory '/home/builder/immortalwrt-mt798x/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#0.81#0.28#2.09
    ERROR: package/feeds/packages/rust [host] failed to build.
make[2]: *** [package/Makefile:118: package/feeds/packages/rust/host/compile] Error 1
Ubuntu20.04 和 22.04编译都会有这个报错

coolsnowwolf, 2026-09-05:
该问题发生在用户手动检出的旧 R23 提交及第三方 kenzok/immortalwrt 组合；当前 feeds/packages Rust 已更新到 1.98.0，不能用旧版 Rust 1.87 的 libc.a 报错判断当前 master。请使用当前 master/官方 feeds，现关闭。

closed 2026-09-05 by coolsnowwolf