coolsnowwolf/lede #13594: JDCloud AX1800 Pro出现 configure: error: libuuid doesn't work properly，编译继续

xinyl189, 2025-07-08:
### 详细叙述

编译京东亚瑟AX1800PRO的时候出现错误，请大佬帮忙看看如何解决，网络已经设置全局了。编译环境是全新的ubuntu24.04。

checking for sysconf... yes
checking for utimensat... yes
checking sysconf (_SC_PAGESIZE)... 4096
checking for pthread.h... yes
checking for pthread_mutex_lock in -lpthread... yes
checking pkg-config is at least version 0.9.0... yes
checking for uuid... yes
checking libuuid usability... no
configure: error: libuuid doesn't work properly
make[3]: *** [Makefile:41: /home/lab/data/lede/build_dir/host/erofs-utils-1.8.9/.configured] Error 1
make[3]: Leaving directory '/home/lab/data/lede/tools/erofs-utils'
time: tools/erofs-utils/compile#22.05#3.74#26.43
    ERROR: tools/erofs-utils failed to build.
make[2]: *** [tools/Makefile:235: tools/erofs-utils/compile] Error 1
make[2]: Leaving directory '/home/lab/data/lede'
make[1]: *** [tools/Makefile:231: /home/lab/data/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnnyyynynyyyyyyyynyyynynyynnynnyyyyyynyyyyyyyyyyynynnyyyyynyyy] Error 2
make[1]: Leaving directory '/home/lab/data/lede'
make: *** [/home/lab/data/lede/include/toplevel.mk:231：world] 错误 2
lab@lab-VMware20-1:~/data/lede$

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

JDCloud AX1800 Pro

### 详细日志

checking how to run the C preprocessor... /home/lab/data/lede/staging_dir/host/bin/gcc -E
checking whether termios.h defines TIOCGWINSZ... no
checking whether sys/ioctl.h defines TIOCGWINSZ... yes
checking for inline... inline
checking for int64_t... yes
checking for size_t... yes
checking for ssize_t... yes
checking for struct stat.st_rdev... yes
checking for struct stat.st_atim... yes
checking for struct stat.st_atimensec... no
checking for uint64_t... yes
checking for /home/lab/data/lede/staging_dir/host/bin/gcc options needed to detect all undeclared functions... none needed
checking whether llseek is declared... no
checking whether lseek64 is declared... yes
checking whether memrchr is declared... yes
checking for backtrace... yes
checking for copy_file_range... yes
checking for fallocate... yes
checking for getrlimit... yes
checking for gettimeofday... yes
checking for lgetxattr... yes
checking for llistxattr... yes
checking for lsetxattr... yes
checking for memset... yes
checking for realpath... yes
checking for lseek64... yes
checking for ftello64... yes
checking for pread64... yes
checking for pwrite64... yes
checking for pwritev... yes
checking for posix_fadvise... yes
checking for fstatfs... yes
checking for sendfile... yes
checking for strdup... yes
checking for strerror... yes
checking for strrchr... yes
checking for strtoull... yes
checking for sysconf... yes
checking for utimensat... yes
checking sysconf (_SC_PAGESIZE)... 4096
checking for pthread.h... yes
checking for pthread_mutex_lock in -lpthread... yes
checking pkg-config is at least version 0.9.0... yes
checking for uuid... yes
checking libuuid usability... no
configure: error: libuuid doesn't work properly
make[3]: *** [Makefile:41: /home/lab/data/lede/build_dir/host/erofs-utils-1.8.9/.configured] Error 1
make[3]: Leaving directory '/home/lab/data/lede/tools/erofs-utils'
time: tools/erofs-utils/compile#22.05#3.74#26.43
    ERROR: tools/erofs-utils failed to build.
make[2]: *** [tools/Makefile:235: tools/erofs-utils/compile] Error 1
make[2]: Leaving directory '/home/lab/data/lede'
make[1]: *** [tools/Makefile:231: /home/lab/data/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnnyyynynyyyyyyyynyyynynyynnynnyyyyyynyyyyyyyyyyynynnyyyyynyyy] Error 2
make[1]: Leaving directory '/home/lab/data/lede'
make: *** [/home/lab/data/lede/include/toplevel.mk:231：world] 错误 2
lab@lab-VMware20-1:~/data/lede$ make menuconfig

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

lab@lab-VMware20-1:~/data/lede$

aiamadeus, 2025-07-08:
你是没有 git pull？https://github.com/coolsnowwolf/lede/commit/70265ac2cadc120b0fb3840b390333110194cb5d

aiamadeus, 2025-07-08:
那建议 `make distclean` 重新编译工具链

qingfeng-1, 2025-08-16:
大佬有解决吗？遇到一样的问题

skymirros, 2025-09-08:
[image]