coolsnowwolf/lede #13601: comit ax18(IPQ60XX)缺少WiFi驱动

fat4-alt, 2025-07-13:
### 详细叙述

新编译的comit ax18(IPQ60XX)默认有选WiFi驱动，但编译完成的固件没有WiFi。
在软件里面安装了IPQ6018 ath11k firmware，没有作用。

comit ax18硬改过1G内存，应该是够用的。

[image]

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

comit ax18

### 详细日志

[image]

fat4-alt, 2025-07-13:
> 虽然可能没改内存的是少数，但这机子无线先天残废，真的有必要开启么

WiFi功能还是不错的，至少在一个小房间内，网速没问题。
请教一下，有什么方法开启WiFi？

aiamadeus, 2025-07-13:
```
diff --git a/package/firmware/ath11k-firmware/Makefile b/package/firmware/ath11k-firmware/Makefile
index 2257313b4..40471102e 100644
--- a/package/firmware/ath11k-firmware/Makefile
+++ b/package/firmware/ath11k-firmware/Makefile
@@ -8,13 +8,13 @@
 include $(TOPDIR)/rules.mk
 
 PKG_NAME:=ath11k-firmware
-PKG_SOURCE_DATE:=2023-03-31
-PKG_SOURCE_VERSION:=a039049a9349722fa5c74185452ab04644a0d351
-PKG_MIRROR_HASH:=ed401e3f6e91d70565b3396139193f7e815f410db93700697205ac8ed1b828c5
+PKG_SOURCE_DATE:=2024-10-14
+PKG_SOURCE_VERSION:=15f050122da5ef5bef2cc8c7c19dfb7f98060a49
+PKG_MIRROR_HASH:=skip
 PKG_RELEASE:=1
 
 PKG_SOURCE_PROTO:=git
-PKG_SOURCE_URL:=https://github.com/quic/upstream-wifi-fw.git
+PKG_SOURCE_URL:=https://git.codelinaro.org/clo/ath-firmware/ath11k-firmware.git
 
 PKG_LICENSE_FILES:=LICENSE.qca_firmware
 
@@ -62,21 +62,10 @@ define Build/Compile
 
 endef
 
-QCN9074_BOARD_REV:=8e140c65f36137714b6d8934e09dcd73cb05c2f6
-QCN9074_BOARD_FILE:=board-2.bin.$(QCN9074_BOARD_REV)
-
-define Download/qcn9074-board
-  URL:=https://github.com/kvalo/ath11k-firmware/raw/master/QCN9074/hw1.0/
-  URL_FILE:=board-2.bin
-  FILE:=$(QCN9074_BOARD_FILE)
-  HASH:=dbf0ca14aa1229eccd48f26f1026901b9718b143bd30b51b8ea67c84ba6207f1
-endef
-$(eval $(call Download,qcn9074-board))
-
 define Package/ath11k-firmware-ipq5018/install
 	$(INSTALL_DIR) $(1)/lib/firmware/ath11k/IPQ5018/hw1.0
 	$(INSTALL_DATA) \
-		$(PKG_BUILD_DIR)/ath11k-firmware/IPQ5018_QCN6122_QCN6122/hw1.0/2.7.0.1/WLAN.HK.2.7.0.1-01744-QCAHKSWPL_SILICONZ-1/*.* \
+		$(PKG_BUILD_DIR)/IPQ5018_QCN6122_QCN6122/hw1.0/2.7.0.1/WLAN.HK.2.7.0.1-01744-QCAHKSWPL_SILICONZ-1/*.* \
 		$(1)/lib/firmware/ath11k/IPQ5018/hw1.0/
 endef
 
@@ -84,14 +73,16 @@ define Package/ath11k-firmware-ipq6018/install
 	$(INSTALL_DIR) $(1)/lib/firmware/IPQ6018
 	$(INSTALL_DIR) $(1)/lib/firmware/ath11k/IPQ6018/hw1.0
 	$(INSTALL_DATA) \
-		$(PKG_BUILD_DIR)/ath11k-firmware/IPQ6018/hw1.0/2.5.0.1/WLAN.HK.2.5.0.1-01208-QCAHKSWPL_SILICONZ-1/* \
+		$(PKG_BUILD_DIR)/IPQ6018/hw1.0/2.5.0.1/WLAN.HK.2.5.0.1-03982-QCAHKSWPL_SILICONZ-3/* \
 		$(1)/lib/firmware/IPQ6018/
+	$(INSTALL_BIN) \
+		$(PKG_BUILD_DIR)/IPQ6018/hw1.0/board-2.bin $(1)/lib/firmware/ath11k/IPQ6018/hw1.0/
 endef
 
 define Package/ath11k-firmware-ipq8074/install
 	$(INSTALL_DIR) $(1)/lib/firmware/IPQ8074
 	$(INSTALL_DATA) \
-		$(PKG_BUILD_DIR)/ath11k-firmware/IPQ8074/hw2.0/testing/2.9.0.1/WLAN.HK.2.9.0.1-01385-QCAHKSWPL_SILICONZ-1/* \
+		$(PKG_BUILD_DIR)/IPQ8074/hw2.0/2.9.0.1/WLAN.HK.2.9.0.1-02146-QCAHKSWPL_SILICONZ-1/* \
 		$(1)/lib/firmware/IPQ8074/
 endef
 
@@ -105,10 +96,10 @@ endef
 define Package/ath11k-firmware-qcn9074/install
 	$(INSTALL_DIR) $(1)/lib/firmware/ath11k/QCN9074/hw1.0
 	$(INSTALL_DATA) \
-		$(PKG_BUILD_DIR)/ath11k-firmware/QCN9074/hw1.0/testing/2.9.0.1/WLAN.HK.2.9.0.1-01385-QCAHKSWPL_SILICONZ-1/* \
+		$(PKG_BUILD_DIR)/QCN9074/hw1.0/2.9.0.1/WLAN.HK.2.9.0.1-02146-QCAHKSWPL_SILICONZ-1/* \
 		$(1)/lib/firmware/ath11k/QCN9074/hw1.0/
 	$(INSTALL_BIN) \
-		$(DL_DIR)/$(QCN9074_BOARD_FILE) $(1)/lib/firmware/ath11k/QCN9074/hw1.0/board-2.bin
+		$(PKG_BUILD_DIR)/QCN9074/hw1.0/board-2.bin $(1)/lib/firmware/ath11k/QCN9074/hw1.0/board-2.bin
 endef
 
 $(eval $(call BuildPackage,ath11k-firmware-ipq5018))
diff --git a/target/linux/qualcommax/files/arch/arm64/boot/dts/qcom/ipq6000-ax18.dts b/target/linux/qualcommax/files/arch/arm64/boot/dts/qcom/ipq6000-ax18.dts
index e887ff87a..b539d8dbb 100644
--- a/target/linux/qualcommax/files/arch/arm64/boot/dts/qcom/ipq6000-ax18.dts
+++ b/target/linux/qualcommax/files/arch/arm64/boot/dts/qcom/ipq6000-ax18.dts
@@ -213,6 +213,6 @@
 };
 
 &wifi {
-	status = "disabled";
+	status = "okay";
 	qcom,ath11k-fw-memory-mode = <2>;
 };
```

fat4-alt, 2025-07-13:
> ```
> diff --git a/package/firmware/ath11k-firmware/Makefile b/package/firmware/ath11k-firmware/Makefile
> index 2257313b4..40471102e 100644
> --- a/package/firmware/ath11k-firmware/Makefile
> +++ [...]

我是小白，太复杂看不懂。
WiFi驱动不是编译进固件了吗？

1980490718, 2025-07-14:
cd lede && touch 1.patch && vi 1.patch
复制上面的内容
shift+insert到里面
ESC：wq!保存退出
git apply 1.patch

fat4-alt, 2025-07-14:
> 复制上面的内容 在lede根目录下打开终端输入 `touch 1.patch && vi 1.patch` 按shift+insert插入到vi窗口里面 分别按ESC：wq!保存退出 `git apply 1.patch` 然后重新编译即可

是在编译固件前的操作，还是编译完固件后的操作？
我每次都是开一个vultr vps来新编译固件，编译完后删除vps