coolsnowwolf/lede #13610: 固件类型选项，如果选erofs不选squashfs，EFI选项消失不可用

loydcheung, 2025-07-14:
### 详细叙述

不确定这个是不是bug，menuconfig下target image菜单中如果不选squashfs，就无法为erofs增加EFI固件的选项。导致生成的固件只有rootfs.img没有combined-efi.img

当然了选中squashfs是可以满足需求，但确实没必要每次两种固件类型都编译。

*我倒是没有试过，更新固件是不是刷不带引导的rootfs.img也可以，知道的大佬可以指点下。谢谢

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

x86

### 详细日志

[image]

[image]

aiamadeus, 2025-07-14:
你试一下只编 erofs 能不能正常启动
```
diff --git a/config/Config-images.in b/config/Config-images.in
index c3d243b51..66b1d3e4b 100644
--- a/config/Config-images.in
+++ b/config/Config-images.in
@@ -217,7 +217,7 @@ menu "Target Images"
 	config GRUB_IMAGES
 		bool "Build GRUB images"
 		depends on TARGET_x86
-		depends on TARGET_ROOTFS_EXT4FS || TARGET_ROOTFS_JFFS2 || TARGET_ROOTFS_SQUASHFS
+		depends on TARGET_ROOTFS_EROFS || TARGET_ROOTFS_EXT4FS || TARGET_ROOTFS_JFFS2 || TARGET_ROOTFS_SQUASHFS
 		select PACKAGE_grub2
 		select PACKAGE_grub2-bios-setup
 		default n
@@ -225,7 +225,7 @@ menu "Target Images"
 	config GRUB_EFI_IMAGES
 		bool "Build GRUB EFI images"
 		depends on TARGET_x86 || TARGET_armsr || TARGET_loongarch64 || TARGET_phytium
-		depends on TARGET_ROOTFS_EXT4FS || TARGET_ROOTFS_JFFS2 || TARGET_ROOTFS_SQUASHFS
+		depends on TARGET_ROOTFS_EROFS || TARGET_ROOTFS_EXT4FS || TARGET_ROOTFS_JFFS2 || TARGET_ROOTFS_SQUASHFS
 		select PACKAGE_grub2 if TARGET_x86
 		select PACKAGE_grub2-efi if TARGET_x86
 		select PACKAGE_grub2-bios-setup if TARGET_x86
```

loydcheung, 2025-07-15:
> 你试一下只编 erofs 能不能正常启动
> ```
> diff --git a/config/Config-images.in b/config/Config-images.in
> index c3d243b51..66b1d3e4b 100644
> --- a/config/Config-images.in
> +++ b/config/Config-images.in
> @@ -217,7 +217,7 [...]

经测试，这样不选squashfs，单独编译出来的erofs-combined-efi一切正常，谢谢。

commit 340b85f3 (2025-07-16): image: add EROFS to GRUB images

Fixes: #13610
closed 2025-07-16 by aiamadeus