coolsnowwolf/lede #13640: 拉取最新源码编译rockchip固件报错

bi4tms, 2025-07-26:
### 详细叙述

拉取最新源码编译rockchip固件报错，提示缺少组件
@coolsnowwolf 

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

r4s

### 详细日志

Error: /home/r4s.6.6/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.6.100/arch/arm64/boot/dts/rockchip/rk3576-photonicat2.dts:348.1-10 Label or path hdptxphy not found
Error: /home/r4s.6.6/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.6.100/arch/arm64/boot/dts/rockchip/rk3576-photonicat2.dts:352.1-6 Label or path hdmi not found
Error: /home/r4s.6.6/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.6.100/arch/arm64/boot/dts/rockchip/rk3576-photonicat2.dts:357.1-9 Label or path hdmi_in not found
Error: /home/r4s.6.6/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.6.100/arch/arm64/boot/dts/rockchip/rk3576-photonicat2.dts:363.1-10 Label or path hdmi_out not found
Error: /home/r4s.6.6/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.6.100/arch/arm64/boot/dts/rockchip/rk3576-photonicat2.dts:749.1-7 Label or path pcie0 not found
Error: /home/r4s.6.6/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.6.100/arch/arm64/boot/dts/rockchip/rk3576-photonicat2.dts:757.1-7 Label or path pcie1 not found
Error: /home/r4s.6.6/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.6.100/arch/arm64/boot/dts/rockchip/rk3576-photonicat2.dts:820.1-12 Label or path pwm0_2ch_0 not found
Error: /home/r4s.6.6/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.6.100/arch/arm64/boot/dts/rockchip/rk3576-photonicat2.dts:827.1-12 Label or path pwm1_6ch_0 not found
Error: /home/r4s.6.6/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.6.100/arch/arm64/boot/dts/rockchip/rk3576-photonicat2.dts:834.1-12 Label or path pwm2_8ch_2 not found
Error: /home/r4s.6.6/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.6.100/arch/arm64/boot/dts/rockchip/rk3576-photonicat2.dts:841.1-6 Label or path sai6 not found
Error: /home/r4s.6.6/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.6.100/arch/arm64/boot/dts/rockchip/rk3576-photonicat2.dts:922.1-5 Label or path vop not found
Error: /home/r4s.6.6/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.6.100/arch/arm64/boot/dts/rockchip/rk3576-photonicat2.dts:926.1-9 Label or path vop_mmu not found
Error: /home/r4s.6.6/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.6.100/arch/arm64/boot/dts/rockchip/rk3576-photonicat2.dts:930.1-5 Label or path vp0 not found
FATAL ERROR: Syntax error parsing input tree
make[5]: *** [Makefile:83: /home/r4s.6.6/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3576-photonicat2.dtb] Error 1
make[5]: Leaving directory '/home/r4s.6.6/lede/target/linux/rockchip/image'
make[4]: *** [Makefile:24: install] Error 2
make[4]: Leaving directory '/home/r4s.6.6/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/r4s.6.6/lede/target/linux'
time: target/linux/install#13.21#4.70#23.18
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/home/r4s.6.6/lede'
make[1]: *** [target/Makefile:24: /home/r4s.6.6/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/r4s.6.6/lede'
make: *** [/home/r4s.6.6/lede/include/toplevel.mk:231：world] 错误 2

bi4tms, 2025-07-26:
删除lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8文件夹，重新编译内核仍然报相同错误

shjy223534, 2025-07-26:
有些rockchip设备已经不支持6.6内核了，你肯定选了6.6内核，选择6.12内核就好了，你应该旋转了使用test内核了，现在test内核变成了6.6,取消test内核就是6.12内核了，所以你只要取消选择test内核就好了

bi4tms, 2025-07-27:
> 有些rockchip设备已经不支持6.6内核了，你肯定选了6.6内核，选择6.12内核就好了，你应该旋转了使用test内核了，现在test内核变成了6.6,取消test内核就是6.12内核了，所以你只要取消选择test内核就好了

你真的确定自己在表达什么吗，现在的代码我编译r2s/r4s/r66s的6.6内核的固件全部报错，而3天前的代码可以正常编译

shjy223534, 2025-07-27:
> > 有些rockchip设备已经不支持6.6内核了，你肯定选了6.6内核，选择6.12内核就可以了，你应该旋转了使用test内核了，现在test内核变成了6.6，取消test内核就是6.12内核了，所以你只要取消选择test内核就好了
> 
> 你真的确定自己在表达什么吗，现在的代码我编译了r2s/r4s/r66s的6.6内核的固件全部报错，而3天前的代码可以正常编译
你这种问题老早遇到过了，别自己不懂以为别人也不懂，你三天前能编译和今天不能编译有什么关系，代码更新了不能编译很正常，你这个问题老早之前就遇到过了，人家lean大佬都解释过了，现在的代码在逐渐放弃6.6内核转向6.12有什么问题么

shjy223534, 2025-07-27:
别人愿意给你解答不感谢就算了，什么垃圾态度，白嫖惯了吧

bi4tms, 2025-07-27:
> 别人愿意给你解答你不感谢就算了，什么垃圾态度，别人欠你么，白嫖习惯了吧

我是在向项目作者提出问题，不需要你的解答，哪儿凉快哪儿待着去

shjy223534, 2025-07-27:
> > 别人愿意给你解答你不感谢就算了，什么垃圾态度，别人欠你么，白嫖习惯了吧
> 
> 我是在向项目作者提出问题，不需要你的解答，也轮不到你来回答，哪儿凉快哪儿待着去

你这样的狗屎人渣还想要人家作者回答，等你全家死光了再说吧，还有你种死全家的就别玩编译了，智商都不够，和小孩去一桌吧

bi4tms, 2025-07-27:
> 这么简单的问题都接不了，issue发这么久了，人家作者连鸟你的心情都没有，还是去看喜羊羊吧，玩什么编译，是你能玩的明白的么

哪个人的裤裆拉链没拉好，把你给放出来了，哪里来滚回哪里去，懒得搭理你。

shjy223534, 2025-07-27:
> > > > 这么简单的问题都接不了，issue发这么久了，人家作者连鸟你的心情都没有，还是去看喜羊羊吧，玩什么编译，是你能玩的明白的么
> > > 
> > > 
> > > 哪个人的裤裆拉链没拉好，把你给放出来了，哪里来滚回哪里去，懒得搭理你。
> > 
> > 
> > 家里死完了记得@我哦
> 
> 把你挂在这里让大家围观脑残

我无所谓啊，只要你全家死了就好的呀

shjy223534, 2025-07-27:
> > > > 这么简单的问题都接不了，issue发这么久了，人家作者连鸟你的心情都没有，还是去看喜羊羊吧，玩什么编译，是你能玩的明白的么
> > > 
> > > 
> > > 哪个人的裤裆拉链没拉好，把你给放出来了，哪里来滚回哪里去，懒得搭理你。
> > 
> > 
> > 家里死完了记得@我哦
> 
> 把你挂在这里让大家围观脑残
我猜是你麻麻没关注逼，乱搞，搞出了你，不然也不会这么傻

shjy223534, 2025-07-27:
死全家了记得@我

bi4tms, 2025-07-27:
静静地注视着脑残

shjy223534, 2025-07-27:
> 静静地注视着脑残

这么快死完啦

closed 2025-07-28 by bi4tms