coolsnowwolf/lede #13642: Github云编译失败，拉取最新源码

Shawleevip, 2025-07-26:
### 详细叙述

内核6.12

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

x86

### 详细日志

 make[3] -C target/linux compile
    ERROR: target/linux failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/runner/work/lede/lede/include/toplevel.mk:231: world] Error 1
WARNING: Makefile 'package/feeds/small/luci-app-ssr-plus/Makefile' has a dependency on 'dns2socks-rust', which does not exist
make[2]: Entering directory '/home/runner/work/lede/lede/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/runner/work/lede/lede/scripts/config'
make[1]: Entering directory '/home/runner/work/lede/lede'
make[2]: Entering directory '/home/runner/work/lede/lede'
make[3]: Entering directory '/home/runner/work/lede/lede/target/linux'
make[4]: Entering directory '/home/runner/work/lede/lede/target/linux/x86'
rm -f /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/localversion
/home/runner/work/lede/lede/scripts/kconfig.pl  + + /home/runner/work/lede/lede/target/linux/generic/config-6.12 /home/runner/work/lede/lede/target/linux/x86/config-6.12 /home/runner/work/lede/lede/target/linux/x86/64/config-6.12 > /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/runner/work/lede/lede/.config >> /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.target
/home/runner/work/lede/lede/scripts/package-metadata.pl kconfig /home/runner/work/lede/lede/tmp/.packageinfo /home/runner/work/lede/lede/.config 6.12 > /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.override
/home/runner/work/lede/lede/scripts/kconfig.pl 'm+' '+' /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.target /dev/null /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.override > /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.set
mv /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.set /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.old
grep -v INITRAMFS /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.old > /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.set
echo "# CONFIG_INITRAMFS_PRESERVE_MTIME is not set" >> /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.set
rm -rf /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/modules
cmp -s /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.set /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.prev || { cp /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.set /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config; cp /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.set /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.prev; }
export MAKEFLAGS= ; [ -d /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/user_headers ] || make -C /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40 KCFLAGS="-fmacro-prefix-map=/home/runner/work/lede/lede/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/runner/work/lede/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Jul 26 18:50:04 2025" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/runner/work/lede/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=6.12.40 CC="x86_64-openwrt-linux-musl-gcc"  INSTALL_HDR_PATH=/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/user_headers headers_install
grep '=[ym]' /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.set | LC_ALL=C sort | /home/runner/work/lede/lede/staging_dir/host/bin/mkhash md5 > /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.vermagic
touch /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.configured
rm -f /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/vmlinux /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/System.map
make -C /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40 KCFLAGS="-fmacro-prefix-map=/home/runner/work/lede/lede/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/runner/work/lede/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Jul 26 18:50:04 2025" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/runner/work/lede/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=6.12.40 CC="x86_64-openwrt-linux-musl-gcc" olddefconfig
make[5]: Entering directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40'
#
# No change to .config
#
make[5]: Leaving directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40'
make -C /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40 KCFLAGS="-fmacro-prefix-map=/home/runner/work/lede/lede/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/runner/work/lede/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Jul 26 18:50:04 2025" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/runner/work/lede/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=6.12.40 CC="x86_64-openwrt-linux-musl-gcc" bzImage modules
make[5]: Entering directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40'
  DESCEND objtool
  INSTALL libsubcmd_headers
Warning: Kernel ABI header at 'tools/arch/x86/include/asm/orc_types.h' differs from latest version at 'arch/x86/include/asm/orc_types.h'
diff -u tools/arch/x86/include/asm/orc_types.h arch/x86/include/asm/orc_types.h
Warning: Kernel ABI header at 'tools/arch/x86/include/asm/insn.h' differs from latest version at 'arch/x86/include/asm/insn.h'
diff -u tools/arch/x86/include/asm/insn.h arch/x86/include/asm/insn.h
Warning: Kernel ABI header at 'tools/arch/x86/lib/insn.c' differs from latest version at 'arch/x86/lib/insn.c'
diff -u tools/arch/x86/lib/insn.c arch/x86/lib/insn.c
  LINK    /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/tools/objtool/objtool
/usr/bin/ld: /home/runner/work/lede/lede/staging_dir/host/lib/libelf.a(elf_compress.o): in function `__libelf_compress':
elf_compress.c:(.text+0x131): undefined reference to `deflateInit_'
/usr/bin/ld: elf_compress.c:(.text+0x20e): undefined reference to `deflate'
[... cut ...]
time: target/linux/compile#1.60#1.00#2.54
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/runner/work/lede/lede'
make[1]: *** [target/Makefile:23: /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/runner/work/lede/lede'
make: *** [/home/runner/work/lede/lede/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

Lasx, 2025-07-27:
相同问题，objtool编译失败。

liwenjie119, 2025-07-27:
相同问题。另外编译upx时也出错
make[4]: *** /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-aarch64_cortex-a53_musl/upx-4.2.4/src: No such file or directory.  Stop.
make[3]: *** [Makefile:64: /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-aarch64_cortex-a53_musl/upx-4.2.4/.built] Error 2
make[3]: Leaving directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/package/lean/upx'
time: package/lean/upx/compile#0.09#0.14#0.19
    ERROR: package/lean/upx failed to build.
make[2]: *** [package/Makefile:116: package/lean/upx/compile] Error 1

binge8, 2025-07-27:
一样的报错，x64报错，armv8没问题

MilesPoupart, 2025-07-27:
x86 6.12最新源码 一样的问题
```
2025-07-27T03:46:04.3698010Z  make[1] target/compile
2025-07-27T03:46:05.3926661Z  make[2] -C target/linux compile
2025-07-27T03:46:20.3996494Z     ERROR: target/linux failed to build.
2025-07-27T03:46:20.4019186Z make -r target/compile: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
2025-07-27T03:46:20.4019959Z make: *** [/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/include/toplevel.mk:231: target/compile] Error 1
2025-07-27T03:46:23.8272642Z make[2]: Entering directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/scripts/config'
2025-07-27T03:46:23.8273053Z make[2]: 'conf' is up to date.
2025-07-27T03:46:23.8274545Z make[2]: Leaving directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/scripts/config'
2025-07-27T03:46:24.8392324Z make[1]: Entering directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt'
2025-07-27T03:46:25.9003143Z make[2]: Entering directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/target/linux'
2025-07-27T03:46:25.9077668Z make[3]: Entering directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/target/linux/x86'
2025-07-27T03:46:25.9389940Z rm -f /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/localversion
2025-07-27T03:46:25.9399496Z /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/scripts/kconfig.pl  + + /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/target/linux/generic/config-6.12 /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/target/linux/x86/config-6.12 /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/target/linux/x86/64/config-6.12 > /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.target
2025-07-27T03:46:25.9628813Z awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/.config >> /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.target
2025-07-27T03:46:25.9660537Z echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.target
2025-07-27T03:46:25.9669814Z echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.target
2025-07-27T03:46:25.9678088Z echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.target
2025-07-27T03:46:25.9685584Z /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/scripts/package-metadata.pl kconfig /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/tmp/.packageinfo /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/.config 6.12 > /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.override
2025-07-27T03:46:26.0126778Z /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/scripts/kconfig.pl 'm+' '+' /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.target /dev/null /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.override > /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.set
2025-07-27T03:46:26.0388571Z mv /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.set /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.old
2025-07-27T03:46:26.0401293Z grep -v INITRAMFS /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.old > /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.set
2025-07-27T03:46:26.0417806Z echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.set
2025-07-27T03:46:26.0425786Z echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.set
2025-07-27T03:46:26.0433502Z echo "# CONFIG_INITRAMFS_PRESERVE_MTIME is not set" >> /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.set
2025-07-27T03:46:26.0440933Z rm -rf /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/modules
2025-07-27T03:46:26.0451453Z cmp -s /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.40/.config.set /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86 [... cut ...]
2025-07-27T03:46:27.3093134Z make[3]: Leaving directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/target/linux/x86'
2025-07-27T03:46:27.3093635Z make[2]: *** [Makefile:11: compile] Error 2
2025-07-27T03:46:27.3093909Z make[2]: Leaving directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/target/linux'
2025-07-27T03:46:27.3095449Z time: target/linux/compile#0.93#0.09#1.41
2025-07-27T03:46:27.3099330Z     ERROR: target/linux failed to build.
2025-07-27T03:46:27.3100910Z make[1]: *** [target/Makefile:30: target/linux/compile] Error 1
```

RORYSSS, 2025-07-27:
好像是下载的时候遇到了 ssl timeout 的问题，下载不全，所以编译有问题。之前没有这样的问题。感觉是 23.05openwrt 源码出现下载问题了。

Lasx, 2025-07-27:
> 好像是下载的时候遇到了 ssl timeout 的问题，下载不全，所以编译有问题。之前没有这样的问题。感觉是 23.05openwrt 源码出现下载问题了。

下载什么东西，我本地编译也是内核编译不成功
哪一块下载不全，我可以离线下来试试

RORYSSS, 2025-07-27:
就是 make menuconfig 之后make -j8 download V=s，看输出里有的部分下载出错了

Lasx, 2025-07-27:
> make -j8 download V=s，看输出里有的部分下载出错了

我这里没有下载出错

Lasx, 2025-07-27:
目前的方法是回滚一下，git checkout 3fda82b52b921dca4cab45323106ef6ef3eacef5

make dirclean再编译就好了

binge8, 2025-07-29:
今天编译的正常了

tonyzzt001, 2025-07-29:
大佬修复了 [798d563](https://github.com/coolsnowwolf/lede/commit/798d563de9b6c68798a3b488dc2d611b51ea2075)

coolsnowwolf, 2025-07-29:
fixed