coolsnowwolf/lede #13649: 最新编译的固件openvpn无法正常运行

dudulongg, 2025-07-30:
### 详细叙述

今天编译完的6.6.100 X86版本测试后发现openvpn无法正常运行，只能回退到我6月初编译的版本才能正常用。大家遇到这个问题了么，我是这个插件的重度用户，希望大佬帮忙解决问题。

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

x86

### 详细日志

和以前编译的各个版本一样的设置，最新编译出来的openvpn就是无法正常运行，恢复到之前版本就没问题

coolsnowwolf, 2025-08-02:
具体是怎样不正常？ 不做谜语人

dudulongg, 2025-08-03:
> 有没有注意到编译完成的文件名称是什么？有没有airoha字样？

[image]
这是编译完成版本的文件名

dudulongg, 2025-08-07:
[image]
在特殊代码这里把这个删除后保存，再填上再保存一下就莫名其妙的好了

0758jian, 2025-11-12:
> 具体是怎样不正常？ 不做谜语人

大佬，我知道不正常的原因，单选上luci-app-openvpn插件编译完成后，在/etc/init.d/opnevpn 不存在执行文件，所以启动不了。这是在日志中查看到，file not found时发现，我就尝试同时加上luci-app-openvpn-server才会有/etc/init.d/openvpn执行文件