coolsnowwolf/lede #13651: ZeroTier问题

mmeiwenlong, 2025-08-01:
### 详细叙述

重启后就这样了
LEDE R25.7.7 / LuCI openwrt-24.10 branch git-25.174.50965-d7e9cc3

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

Intel(R) Client Systems NUC11TNHi5/NUC11TNBi5 - 11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz : 4C8T(CpuMark : 80099.790990 Scores)

### 详细日志

[image]

这咋搞

coolsnowwolf, 2025-08-02:
配置不兼容的问题

mmeiwenlong, 2025-08-02:
> 配置不兼容的问题
这咋解决，这个不兼容fw4吧

GreatMichaelLee, 2025-08-21:
这个是我目前最头大的问题，因为我的planet是我自己定制的国内云主机，我的解决办法是把以前的/etc/init.d/zerotier 脚本考过来解决了，现在24.10里那个init配置脚本直接就是直接用默认配置起sample_config，根本不拉/etc/config/zero下的配置，这个不行。

mmeiwenlong, 2025-08-22:
能把你这个文件传上来么，我找不到旧的了> 这个是我目前最头大的问题，因为我的planet是我自己定制的国内云主机，我的解决办法是把以前的/etc/init.d/zerotier 脚本考过来解决了，现在24.10里那个init配置脚本直接就是直接用默认配置起sample_config，根本不拉/etc/config/zero下的配置，这个不行。

GreatMichaelLee, 2025-08-22:
> 能把你这个文件传上来么，我找不到旧的了> 这个是我目前最头大的问题，因为我的planet是我自己定制的国内云主机，我的解决办法是把以前的/etc/init.d/zerotier 脚本考过来解决了，现在24.10里那个init配置脚本直接就是直接用默认配置起sample_config，根本不拉/etc/config/zero下的配置，这个不行。

#!/bin/sh /etc/rc.common

START=99

USE_PROCD=1

PROG=/usr/bin/zerotier-one
CONFIG_PATH=/var/lib/zerotier-one

service_triggers() {
        procd_add_reload_trigger "zerotier"
        procd_add_interface_trigger "interface.*.up" wan /etc/init.d/zerotier restart
}

section_enabled() {
        config_get_bool enabled "$1" 'enabled' 0
        [ $enabled -gt 0 ]
}

start_instance() {
        local cfg="$1"
        local port secret config_path
        local ARGS=""

        if ! section_enabled "$cfg"; then
                echo "disabled in config"
                return 1
        fi

  [ -d /etc/config/zero ] || mkdir -p /etc/config/zero
  config_path=/etc/config/zero

        config_get_bool port $cfg 'port'
        config_get secret $cfg 'secret'

        # Remove existing link or folder
        rm -rf $CONFIG_PATH

        # Create link from CONFIG_PATH to config_path
        if [ -n "$config_path" -a "$config_path" != $CONFIG_PATH ]; then
                if [ ! -d "$config_path" ]; then
                        echo "ZeroTier config_path does not exist: $config_path"
                        return
                fi

                ln -s $config_path $CONFIG_PATH
        fi

        mkdir -p $CONFIG_PATH/networks.d

        if [ -n "$port" ]; then
                ARGS="$ARGS -p$port"
        fi

        if [ "$secret" = "generate" ]; then
                echo "Generate secret - please wait..."
                local sf="/tmp/zt.$cfg.secret"

                zerotier-idtool generate "$sf" > /dev/null
                [ $? -ne 0 ] && return 1

                secret="$(cat $sf)"
                rm "$sf"

                uci set zerotier.$cfg.secret="$secret"
                uci commit zerotier
        fi

        if [ -n "$secret" ]; then
                echo "$secret" > $CONFIG_PATH/identity.secret
                # make sure there is not previous identity.public
                rm -f $CONFIG_PATH/identity.public
        fi

        add_join() {
                # an (empty) config file will cause ZT to join a network
                touch $CONFIG_PATH/networks.d/$1.conf
        }

        config_list_foreach $cfg 'join' add_join

        procd_open_instance
        procd_set_param command $PROG $ARGS $CONFIG_PATH
        procd_set_param stderr 1
        procd_close_instance
}

start_service() {
        config_load 'zerotier'
        config_foreach start_instance 'zerotier'
        touch /tmp/zero.log && /etc/zerotier.start > /tmp/zero.log 2>&1 &
}

stop_instance() {
  rm -f /tmp/zero.log
        local cfg="$1"

        /etc/zerotier.stop > /tmp/zero.log 2>&1 &

        # Remove existing link or folder
        rm -f $CONFIG_PATH/networks.d/*.conf
        rm -rf $CONFIG_PATH
}

stop_service() {
        config_load 'zerotier'
        config_foreach stop_instance 'zerotier'
}

reload_service() {
        stop
        start
}

coolsnowwolf, 2026-09-05:
该问题对应旧版 ZeroTier init 脚本没有正确使用 `/etc/config/zerotier` 配置路径。当前 feeds/packages 的 zerotier 已在 commit d9e16404 更新，支持 global 与 per-network 配置、`config_path`/软链接及迁移逻辑；issue 中的旧脚本已不再适用。

closed 2026-09-05 by coolsnowwolf