coolsnowwolf/lede #13680: 防火墙的Fullcone NAT高性能模式和TurboACC 的NAT模式设置冲突

GreatMichaelLee, 2025-08-27:
### 详细叙述

版本最新基线， LuCI openwrt-23.05 branch (git-25.226.54199-44b4855) / LEDE R25.8.8

TurboACC只有兼容模式和禁用选，ACC设为兼容模式就把防火墙的高性能模式给冲掉了，变成了兼容模式，兼容模式NAT是端口限制型的，手工改防火墙配置为高性能可以好，但是重启路由或者保留配置upgrade固件之后又会变成兼容模式，又得手工调防火墙配置为高性能，这个问题痛苦死了，啥时候能修一下啊？

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

x86

### 详细日志

没有日志，就是兼容性问题。

kleisth, 2025-08-28:
这应该不是冲突的问题，我两边都是兼容模式，但测试nat仍旧是全锥型fullcone。

GreatMichaelLee, 2025-08-28:
> 这应该不是冲突的问题，我两边都是兼容模式，但测试nat仍旧是全锥型fullcone。

我只要设置成兼容模式，P**N设备上就是端口限制型，设成高性能立马就是全锥形，立竿见影呢。

kleisth, 2025-08-28:
[image]

[image]

你最好确认下是不是你自己的问题

GreatMichaelLee, 2025-08-29:
不懂代码，只能问AI,我也不知道它是不是胡说八道，大佬看看是不是。我的TurboACC里确实只有两个，是不是UI逻辑根据模块挡掉了？

[image]

[image]

[image]

[image]

GreatMichaelLee, 2025-08-29:
麻烦大佬看看是不是有这个改动引入的

44b4855f4e09348ef16b4aeaa5c9e842ee6e1888

zss12345678-a, 2025-09-29:
> 按AI说的改了，的确好了。
> 
> <img alt="图像" width="2000" height="1442" src="https://private-user-images.githubusercontent.com/20067401/483420972-952aa484-ae2f-4375-a711-c8f70f8163a8.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTkxMTI2OTMsIm5iZiI6MTc1OTExMjM5MywicGF0aCI6Ii8yMDA2NzQwMS80ODM0MjA5NzItOTUyYWE0ODQtYWUyZi00Mzc1LWE3MTEtYzhmNzBmODE2M2E4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTI5VDAyMTk1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM0YWYyYmQ5Y2JjZTAyMDg2ZjFmY2IwMDM2YzEwMGM0NjA2ZjFkNzQyMGFmZWVmNTZmZTRjNmVlYWZmNTI2ZWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.lxa_vVWsqvcwprrT4QuK2C-AX6O8OrpYRqZ5NcHDV14">
> 另外：麻烦大佬看看是不是有这个改动引入的，但感觉这个事防火墙的又不太像，两边msg str对应的太挠头了....
> 
> [44b4855f4e09348ef16b4aeaa5c9e842ee6e1888](https://github.com/coolsnowwolf/luci/commit/44b4855f4e09348ef16b4aeaa5c9e842ee6e1888)

兄弟，遇到了一样的问题，但是不需要你这么复杂，最新的tubroacc加了nft_fullcone,如果检测到这个模块就会自动屏蔽高性能模式，，如果你已经编译好系统在使用了，那么你只需要去把nft_fullcone.ko这个模块删除即可，然后重启一下tubroacc，然后界面一切就显示正常了

coolsnowwolf, 2026-09-05:
当前 TurboACC RPC 代码已经检测 `nft_fullcone.ko` 并据此隐藏不适用的高性能/兼容模式选项；TurboACC 与 firewall4 的 fullcone 实现是不同后端，不能同时由两个配置项接管。issue 后续已给出删除不需要的 `nft_fullcone.ko` 作为 workaround，当前没有足够证据表明 LEDE 仍有可复现源码回归。

closed 2026-09-05 by coolsnowwolf