coolsnowwolf/lede #13709: R5S LTS 最新代码编译固件 LAN 口无法识别

CsoWhy, 2025-09-17:
### 详细叙述

现象和 R4S 的 LAN 无法识别非常类似 WAN 可以识别
通过临时修改 02_network 把 WAN 和 LAN 对换后进入系统看日志 PCIe 有错误日志

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

NanoPi R5S LTS

### 详细日志

----

aiamadeus, 2025-09-17:
没有任何日志？这里支持的是 R5S 而不是 R5S LTS

kleisth, 2025-09-21:
> > 没有任何日志？这里支持的是 R5S 而不是 R5S LTS
> 
> 和 R4S 日志几乎一致，看看有什么参考价值吗？之前的R4S问题贴没解决就关掉了。
> 
> [ 1.040634] rockchip-pcie f8000000.pcie: host bridge /pcie@f8000000 ranges: [ 1.041335] rockchip-pcie f8000000.pcie: MEM 0x00fa000000..0x00fbdfffff -> 0x00fa000000 [ 1.042087] rockchip-pcie f8000000.pcie: IO 0x00fbe00000..0x00fbefffff -> 0x00fbe00000 [ 1.043252] rockchip-pcie f8000000.pcie: no vpcie12v regulator found [ 1.544179] rockchip-pcie f8000000.pcie: PCIe link training gen1 timeout! [ 1.544872] rockchip-pcie: probe of f8000000.pcie failed with error -110

这些日志信息显示了 Linux 系统启动时，Rockchip 平台上 PCIe 控制器（f8000000.pcie）初始化失败的过程，主要问题集中在PCIe 链路训练超时和缺少电源调节器。以下是具体分析和可能的解决方向：
日志关键信息解读
地址范围配置（正常）：前 3 行显示 PCIe 主机桥正确配置了内存（MEM）和 IO 地址范围，这部分初始化正常。
电源调节器缺失（警告）：no vpcie12v regulator found说明系统未找到 PCIe 设备所需的 12V 电源调节器（vpcie12v）。PCIe 设备通常需要特定电压供电，缺少该调节器可能导致设备供电不足，进而影响后续初始化。
链路训练超时（错误核心）：PCIe link training gen1 timeout!PCIe 设备初始化时需要通过 “链路训练” 建立与控制器的连接（gen1指 PCIe 第一代速率，2.5GT/s）。超时表明设备与控制器未能成功建立通信链路。
初始化失败：probe of f8000000.pcie failed with error -110最终 PCIe 控制器初始化失败，错误码-110对应ETIMEDOUT（超时），由链路训练失败导致。
可能的原因及解决思路
硬件连接问题
检查 PCIe 设备（如网卡、SSD 等）是否正确插入插槽，是否存在接触不良。
尝试重新插拔设备或更换插槽（如有多个）。
电源供应问题
若硬件设计中确实存在 12V PCIe 电源（vpcie12v），可能是电源调节器故障或未使能，需检查硬件供电电路。
若硬件无vpcie12v（如部分嵌入式设备简化设计），需修改设备树（Device Tree），删除对该调节器的依赖配置（例如注释掉vpcie12v-supply节点）。
驱动或设备树配置错误
检查设备树中 PCIe 控制器节点（/pcie@f8000000）的配置，确保地址范围、时钟、复位等参数与硬件匹配。
更新内核或 Rockchip PCIe 驱动至最新版本，可能修复链路训练的兼容性问题。
速率兼容性问题部分设备可能在高世代速率（如 gen2/gen3）下训练失败，可尝试在设备树中强制 PCIe 工作在 gen1 模式：
dts
&pcie {
    max-link-speed = <1>;  // 强制gen1速率
    // 其他配置...
};
固件或硬件兼容性
更新主板 BIOS/UEFI 固件（如有），确保对 PCIe 控制器的支持。
确认 PCIe 设备与 Rockchip 平台的兼容性（部分设备可能需要特定初始化序列）。
总结
核心问题是 PCIe 链路训练失败，可能由电源配置不当、硬件连接问题或驱动 / 设备树错误导致。建议优先检查硬件连接和电源配置，再逐步排查设备树和驱动兼容性。

CsoWhy, 2025-09-26:
> 这个问题可能是从内核切到`6.12`开始的，3/3B+也一样`LAN`口起不来，手上的树莓派还是用的 `6.6` 版本编译的

我切换到 6.6 内核发现也不行 LAN 口启动不了 肯能是 6.x 的问题

coolsnowwolf, 2025-09-30:
测试加入 pcie 初始化 delay 是否能解决，你们日志太不全了

[image]

BenjaminX, 2025-10-07:
> 测试加入 pcie 初始化 delay 是否能解决，你们日志太不全了
> 
> <img alt="Image" width="1312" height="646" src="https://private-user-images.githubusercontent.com/31687149/495512865-32dbc9a9-4b11-4045-9e9a-dd0f929f15e3.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTk4Mzg1MDMsIm5iZiI6MTc1OTgzODIwMywicGF0aCI6Ii8zMTY4NzE0OS80OTU1MTI4NjUtMzJkYmM5YTktNGIxMS00MDQ1LTllOWEtZGQwZjkyOWYxNWUzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTEwMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMDA3VDExNTY0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBlZjZhMGFiOWU2MTkzNzEwYzlhZWQ1NGQ0ZDFkZThjMmJjZWVmMjI4ZmZiZmY3YzY3ZjZmYzJhYzc4MDhhYWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.o-vsr0e-ABYbeedr94nWHX-p8vsr7m25MzVeqPUZi68">

还是不行 LAN 拉不起来。

coolsnowwolf, 2026-09-05:
当前 NanoPi R5S 支持与 OpenWrt 主线一致，网络映射为 eth1/eth2 LAN、eth0 WAN。issue 后续日志显示的是 f8000000 PCIe 链路训练超时（-110），但未给出 R5S LTS 的完整 DTS/硬件修订、所有 PCIe 控制器日志及 reset/供电状态；R5S 与 R5S LTS 也可能不是同一硬件。没有这些证据无法安全移植 PCIe delay 或重置补丁，当前源码暂不修改。

closed 2026-09-05 by coolsnowwolf